pyrtlsdr/pyrtlsdr #28: Memorry Error when read_samples > 22*2048000 samples (direct write out possible?)

hornig, 2015-08-29:
Hi,

I encountered Memorry Errors when I do sdr.read_samples(22*2048000)).
It doesn't like the complex type with Python 2.7.10.
Is there a solution to circumvent this, like a direct write out to file?
The rtl_sdr allows it, but I want to avoid using it directly because I need to have a small as possible sample lost between read/write outs.

I don't know if you also have this error or of this is already known (couldn't find it here).
And unfortunately, I need to work with these big chunks of data and even bigger.

Thanks for PyRTLSDR! :)

Best, Andreas

michelp, 2015-08-29:
Do you have the complete traceback?  I'm not sure what "It doesn't like the complex type" means.  A MemoryError in Python usually means your process has run out of memory.

roger-, 2015-08-29:
Hi,

> It doesn't like the complex type with Python 2.7.10.

Can you elaborate on this?

> Is there a solution to circumvent this, like a direct write out to file?

Assuming you really need your samples to be contiguous, you have a couple options:

1. use `read_bytes()` which avoids copying/converting the data and should save you some memory. Additionally it's a more space efficient/simpler way to store the data in a file. 

1. Another approach is to use the asynchronous support (e.g. `read_bytes_async()`) and have a callback function that writes the samples. Assuming everything works properly, this would let you write as many samples as you want to a file and probably is the right way to do this.

Regards

nocarryr, 2016-09-04:
@hornig  Has there been any progress here?  I'm going to close this issue, but will re-open if there's still more to discuss.

Thanks!

closed 2016-09-04 by nocarryr

quantenProjects, 2021-01-06:
https://github.com/aerospaceresearch/CalibrateSDR/issues/1

Is this related?

nocarryr, 2021-01-07:
@quantenProjects  I can't say that it isn't, but looking through that repo, I do see something that may be causing the issue you referenced.  I'll comment there.