pyrtlsdr/pyrtlsdr #30: Importing rtlsdraio.py in PY2 causes syntax errors

nocarryr, 2015-10-03:
The changes to method/function syntax in PY3 cause errors on import in PY2:
https://github.com/roger-/pyrtlsdr/blob/master/rtlsdr/rtlsdraio.py#L15

roger-, 2015-10-03:
Ugh, didn't have a 32 bit Python 2 installation so I couldn't test it. I thought [this](https://github.com/roger-/pyrtlsdr/blob/master/rtlsdr/__init__.py#L28) would cover any import errors for Python < 3.5.

nocarryr, 2015-10-03:
Also:
https://github.com/roger-/pyrtlsdr/blob/master/rtlsdr/rtlsdraio.py#L59

Fails on PY < 3.5 during import

nocarryr, 2015-10-03:
> I thought [this](https://github.com/roger-/pyrtlsdr/blob/master/rtlsdr/__init__.py#L28) would cover any import errors for Python < 3.5.

Yeah, I saw that too and would've imagined that'd cover it.  Syntax in the module itself though killed it.  :(

PR #31 (2015-10-03): Avoid importing rtlsdraio in PY2
Fixes #30
closed 2015-10-03 by roger-