pyrtlsdr/pyrtlsdr #35: Syntax error during installation (Python 3)

MaineTim, 2016-02-25:
(pyrtlsdr) shack pyrtlsdr # python setup.py install
running install
running bdist_egg
running egg_info
creating pyrtlsdr.egg-info
writing top-level names to pyrtlsdr.egg-info/top_level.txt
writing pyrtlsdr.egg-info/PKG-INFO
writing dependency_links to pyrtlsdr.egg-info/dependency_links.txt
writing manifest file 'pyrtlsdr.egg-info/SOURCES.txt'
reading manifest file 'pyrtlsdr.egg-info/SOURCES.txt'
writing manifest file 'pyrtlsdr.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib
creating build/lib/rtlsdr
copying rtlsdr/testutils.py -> build/lib/rtlsdr
copying rtlsdr/rtlsdraio.py -> build/lib/rtlsdr
copying rtlsdr/rtlsdrtcp.py -> build/lib/rtlsdr
copying rtlsdr/librtlsdr.py -> build/lib/rtlsdr
copying rtlsdr/__init__.py -> build/lib/rtlsdr
copying rtlsdr/rtlsdr.py -> build/lib/rtlsdr
copying rtlsdr/helpers.py -> build/lib/rtlsdr
creating build/bdist.linux-x86_64
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/rtlsdr
copying build/lib/rtlsdr/testutils.py -> build/bdist.linux-x86_64/egg/rtlsdr
copying build/lib/rtlsdr/rtlsdraio.py -> build/bdist.linux-x86_64/egg/rtlsdr
copying build/lib/rtlsdr/rtlsdrtcp.py -> build/bdist.linux-x86_64/egg/rtlsdr
copying build/lib/rtlsdr/librtlsdr.py -> build/bdist.linux-x86_64/egg/rtlsdr
copying build/lib/rtlsdr/__init__.py -> build/bdist.linux-x86_64/egg/rtlsdr
copying build/lib/rtlsdr/rtlsdr.py -> build/bdist.linux-x86_64/egg/rtlsdr
copying build/lib/rtlsdr/helpers.py -> build/bdist.linux-x86_64/egg/rtlsdr
byte-compiling build/bdist.linux-x86_64/egg/rtlsdr/testutils.py to testutils.cpython-34.pyc
byte-compiling build/bdist.linux-x86_64/egg/rtlsdr/rtlsdraio.py to rtlsdraio.cpython-34.pyc
  File "build/bdist.linux-x86_64/egg/rtlsdr/rtlsdraio.py", line 68
    async def __aiter__(self):
            ^
SyntaxError: invalid syntax

byte-compiling build/bdist.linux-x86_64/egg/rtlsdr/rtlsdrtcp.py to rtlsdrtcp.cpython-34.pyc
byte-compiling build/bdist.linux-x86_64/egg/rtlsdr/librtlsdr.py to librtlsdr.cpython-34.pyc
byte-compiling build/bdist.linux-x86_64/egg/rtlsdr/__init__.py to __init__.cpython-34.pyc
byte-compiling build/bdist.linux-x86_64/egg/rtlsdr/rtlsdr.py to rtlsdr.cpython-34.pyc
byte-compiling build/bdist.linux-x86_64/egg/rtlsdr/helpers.py to helpers.cpython-34.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying pyrtlsdr.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pyrtlsdr.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pyrtlsdr.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pyrtlsdr.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/pyrtlsdr-0.1.1-py3.4.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing pyrtlsdr-0.1.1-py3.4.egg
Copying pyrtlsdr-0.1.1-py3.4.egg to /data/usr/local/src/pyrtlsdr/lib/python3.4/site-packages
Adding pyrtlsdr 0.1.1 to easy-install.pth file
Installed /data/usr/local/src/pyrtlsdr/lib/python3.4/site-packages/pyrtlsdr-0.1.1-py3.4.egg
Processing dependencies for pyrtlsdr==0.1.1
Finished processing dependencies for pyrtlsdr==0.1.1
(pyrtlsdr) shack pyrtlsdr # python --version
Python 3.4.3
 
This happens both on a Linux Mint PC and a Raspberry Pi running Raspbian Jessie.

MaineTim, 2016-02-25:
Ah, sorry I didn't read the README all the way through before posting. Python 3.5+ required for that...

closed 2016-02-25 by MaineTim

nocarryr, 2016-02-25:
I'm still curious though, the setup script should be ignoring the `rtlsdraio` module if it detects that the python version is less than 3.5.  Was this from a recent download off of the github repo?

MaineTim, 2016-02-26:
Looks like you've already cleared this one up, but yes, I cloned it moments before compiling.

nocarryr, 2016-02-26:
Yeah, I got it wrong initially with the packaging script. Hopefully those errors won't be an issue anymore.

roger-, 2016-02-26:
I've been thinking that we could probably avoid the Python 3 errors (from `await`/`async`) by just doing a dynamic import like:

```python
from  rtlsdr import RtlSdr

try:
    import importlib
    
    RtlSdr = importlib.import_module('.rtlsdraio').RtlSdrAio    
except ImportError:
    pass
```

Haven't tested it yet, but I believe this way Python 2 won't try to parse the aio stuff and make things a bit simpler.

nocarryr, 2016-02-26:
The problem though is that setuptools/distutils still sees the .py file and tries to byte-compile it as part of the setup command.  The only other way I can think of to avoid this would be to nest it in its own package, but that seems a bit odd for just one module.

roger-, 2016-02-26:
Oh right. How about just disabling byte-compiling then?

nocarryr, 2016-02-26:
It could be done that way I suppose.  The only thing would be if in the future you wanted to update the release on PyPI.  You'd have the same issue since a bdist_wheel would have to be generated (for py2 and py3.5+)

roger-, 2016-02-26:
Ah, didn't realize that.

commit e3702ac9 (2016-02-25): Subclass distutils commands to exclude py3.5-only modules
Fixes #35