pyrtlsdr/pyrtlsdr #40: pip install error on RPi3

MatejKovacic, 2016-04-12:
I try to install it ba:
sudo pip install pyrtlsdr

However, I got this:
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python2.7/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
    from pip.download import path_to_url
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 25, in <module>
    from requests.compat import IncompleteRead
ImportError: cannot import name IncompleteRead

nocarryr, 2016-04-12:
Do you know what version of pip you have? Try updating it to see if that changes anything:
`sudo pip install -U pip`

MatejKovacic, 2016-04-13:
Hmm, this:
`sudo pip install -U pip`

Gives me:
```
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python2.7/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
    from pip.download import path_to_url
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 25, in <module>
    from requests.compat import IncompleteRead
ImportError: cannot import name IncompleteRead

```

nocarryr, 2016-04-13:
Seems like the cause is from either an outdated version of pip or the 'requests' library. What version of pip is installed?
`pip --version ` 
Also, what distro are you running on your RPI? Are you up to date with whatever package manager it comes with?

MatejKovacic, 2016-04-13:
Sorry, forgot to mention.

I am using Raspbian Jessie 8.0. Up-to-date (apt-get update && apt-get upgrade).

However, this is weird:
`pip --version`

I got this:
```
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python2.7/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
    from pip.download import path_to_url
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 25, in <module>
    from requests.compat import IncompleteRead
ImportError: cannot import name IncompleteRead

```

It seems something is seriously broken here...

MatejKovacic, 2016-04-13:
Anyway, when I try to:
`sudo apt-get upgrade`

It says:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  pypy-upstream raspberrypi-ui-mods
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
```

nocarryr, 2016-04-13:
Wow, yeah I would say so.
Maybe if you tried updating pip using 'easy_install':

`sudo easy_install -U pip'

Or if that doesn't work, you may be able to install pyrtlsdr by using easy_install instead of pip, but I don't know if it'll install properly.

nocarryr, 2016-04-13:
Ok, from your last comment... Did you call `sudo apt-get update` first?

MatejKovacic, 2016-04-13:
Yes, of course.

Anyway,  I said sudo easy_install -U pip nad now I got:
```
pip --version
pip 1.5.6 from /usr/lib/python2.7/dist-packages (python 2.7)

```

And `sudo pip install pyrtlsdr` is also working now. Thanks for your help.

nocarryr, 2016-04-13:
No problem!

closed 2016-04-13 by nocarryr