pyrtlsdr/pyrtlsdr #41: I guess one

kornos, 2016-04-28:
Hello 

Yesterday i installed pyrtlsdr from github.
I tried to run your test.py, but it fails with: 

:~/projects/python$ python test.py 
Traceback (most recent call last):
  File "rtlsdr3.py", line 21, in <module>
    from rtlsdr.testutils import (
ImportError: No module named testutils

If I start python console :
~/projects/python$ python
Python 2.7.10 (default, Oct 14 2015, 16:09:02) 
[GCC 5.2.1 20151010] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> dir()
['__builtins__', '__doc__', '__name__', '__package__']
>>> import rtlsdr
>>> dir()
['__builtins__', '__doc__', '__name__', '__package__', 'rtlsdr']
>>> dir(rtlsdr)
['__builtins__', '__doc__', '__file__', '__name__', '__package__', 'itertools', 'math', 'matplotlib', 'rtlsdr', 'threading', 'time']

import rtlsdr  it works , but RtlSdr() is nowhere to be found.
What is wrong here?

RtlSdrScaner ,which uses pyrtlsdr , works OK on my system however.

Thanks
Victor

nocarryr, 2016-04-28:
That module was originally in place before more comprehensive tests were put in place.  Those files won't typically be installed from a normal installation through pip though.

I apparently forgot to update the original `test.py` with the most recent changes though.  I can see how it would be useful for a fresh install when you just want to make sure everything is working.

Thanks for the report!

nocarryr, 2016-04-28:
@kornos I made the corrections, but when testing on one of my machines here, I got an error during the `read_samples_async` portion.  It was not an error generated by the python wrapper, but something from the driver.

It could just be my setup here though since I haven't played with my sdr for a while.  Could you let me know if you see anything?

closed 2016-04-28 by nocarryr

kornos, 2016-04-28:
@nocarryr : Thanks for you quick response.

When running test.py now , I get:

kornos@ubuntu:~/projects/python$ python test.py 
Traceback (most recent call last):
  File "test.py", line 62, in <module>
    main()
  File "test.py", line 24, in main
    @limit_calls(2)
NameError: global name 'limit_calls' is not defined

nocarryr, 2016-04-28:
That may be because of the import path.  It's expecting to be run from within the project root, so if you did a clone of pyrtlsdr under `~/projects/python`, you should be calling it from within the directory `~/projects/python/pyrtlsdr`.

Or I could be misreading your shell prompt

kornos, 2016-04-28:
Yes , you are right.
When I run test.py from project root , it runs OK.

But I have a question:
Let's take this small example:

kornos@ubuntu:~/projects/python$ cat rtlsdr2.py 
from rtlsdr import RtlSdr

sdr = RtlSdr()

# configure device
sdr.sample_rate = 2.048e6  # Hz
sdr.center_freq = 70e6     # Hz
sdr.freq_correction = 60   # PPM
sdr.gain = 'auto'

print(sdr.read_samples(512))

When i run it (outside project root) , it spews this error, although pyrtlsdr in installed globally:

kornos@ubuntu:~/projects/python$ python rtlsdr2.py 
Traceback (most recent call last):
  File "rtlsdr2.py", line 1, in <module>
    from rtlsdr import RtlSdr
ImportError: cannot import name RtlSdr

Why is that?
I have to run all my .py files in pyrtlsdr project root?

Thanks
Victor

nocarryr, 2016-04-28:
This is happening because you're running a script one directory up from where you have the project cloned.  There's sort of a weird namespace issue with the layout:
`rtlsdr/rtlsdr.py` as a dotted import would be `rtlsdr.rtlsdr`

It's something that's been on my mind to bring up, but for the most part, it doesn't cause issues.  If you have it installed globally (I'm assuming through `setup.py install`) then you should be able to import as you did in your script, but it would need to be further away from where you cloned the repo.  Something like:
`~/projects/python/mysdrscripts/sdr2.py`

kornos, 2016-04-28:
Damn , you're right again :-D

So , I have my pyrtlsdr project root in : 
kornos@ubuntu:~/projects/SDR/pyrtlsdr$ pwd
/home/kornos/projects/SDR/pyrtlsdr

and my rtlsdr2.py script in 
kornos@ubuntu:~/projects/python$ pwd
/home/kornos/projects/python

It doesn't work now but if I make a new dir /project/python/myscripts
and run it from here , it works, as you said.

Well , this is weird :-D , if there is a way maybe you can fix it , it drew me mad for the last 2 days.

Thanks very much
Victor

nocarryr, 2016-04-28:
Yeah, import paths can be a real beating sometimes.  Glad we found a solution though.

commit a090a845 (2016-04-28): Fixes #41 - Correct import errors in test.py