pyrtlsdr/pyrtlsdr #42: Pip install pyrtlsdr error

kornos, 2016-04-28:
This morning I tried to install pyrtlsdr via pip install, I got this error:

kornos@ubuntu:~/projects/SDR/pyrtlsdr$ sudo pip install pyrtlsdr
Downloading/unpacking pyrtlsdr
  Downloading pyrtlsdr-0.2.2-py2-none-any.whl
Installing collected packages: pyrtlsdr
Compiling /tmp/pip-build-1mYngd/pyrtlsdr/rtlsdr/rtlsdraio.py ...
  File "/tmp/pip-build-1mYngd/pyrtlsdr/rtlsdr/rtlsdraio.py", line 24
    def __init__(self, func_start, func_stop=None, queue_size=20, *, loop=None):
                                                                   ^
SyntaxError: invalid syntax

Successfully installed pyrtlsdr
Cleaning up...

Running RtlSdrScanner after this produced all kind of weird errors.(segfaults,etc)
Pip uninstall and after that git clone pyrtlsdr fixed it.

Sorry if I report in the wrong place , but I think pip install rtlsdr is flawed.

Thanks
Victor

nocarryr, 2016-04-28:
That's an issue with a module meant for a later python version (3.5).  It uses keywords that don't exist in older versions so it's *supposed* to be excluded from packages for the incompatible versions.

I checked the wheel file (`pyrtlsdr-0.2.2-py2-none-any.whl`) and verified that it somehow made its way back in there so I'll have do some digging and figure out why.

Thanks for bringing this up

kornos, 2016-04-28:
Wow, you must be the fastest bugfixer I saw so far :-O

Best regards
Victor

nocarryr, 2016-04-28:
Lol, putting off doing yard work. :laughing:

nocarryr, 2016-04-30:
Should be fixed from PR #43 and now released to PyPI as v0.2.3

@kornos Let me know if that didn't work for you

closed 2016-04-30 by nocarryr

kornos, 2016-04-30:
Yep , it works like a charm now.

Thanks very much.