pyrtlsdr/pyrtlsdr #44: Make rtlsdrtcp client not need librtlsdr to run.

chibill, 2016-07-02:
Currently the Python top client needs the c lib because it loads up the main Python file but from looking at the code it seems it never actually used the lib which means you could make it so the client does not need it.

nocarryr, 2016-07-02:
I agree.  Loading the wrapper library is unnecessary and to be honest, the original goal for the whole tcp module was to be able to use rtlsdr on a device/os that may not be able to compile it (namely Android).

But since it inherits from the core `RtlSdr` class, the `rtlsdr.py` module has to be loaded which in turn imports the wrapper lib.

I suppose the easiest way to make this work would be to split the tcp module into a package with the client, server and message-related items in their own modules.  The client class overrides pretty much everything from the base class so it wouldn't take too much work to do this.

The remaining issue would then be the [top-level imports](https://github.com/roger-/pyrtlsdr/blob/master/rtlsdr/__init__.py#L18) for the entire library.  The wrapper is the first module to be loaded (and gets imported again through the other modules).  I'm not sure what the most graceful solution is to that.

If we silently ignore the missing lib at import time:
```python
try:
    from librtlsdr import librtlsdr
except ImportError:
    from .librtlsdr import librtlsdr
except ImportError:
    pass#?
```
That could throw people off since they wouldn't know whether or not they've successfully installed everything.

I'm not raising these concerns to say it can't be done (I too would like to see this be possible), just thinking out loud and trying to figure out the best way to go about it.

chibill, 2016-07-02:
You could have the wrapper have a fuction that needs to be called to load the wrapper. And the reason I was asking about this was because I wanted to use it on my iPad in pythonista

chibill, 2016-07-24:
Another sollution would be to have it print out "Unable to import librtlsdr. Hardware based functionality will not work."

nocarryr, 2016-07-24:
Thanks for the bump. I forgot about this. Will start looking into it.
I think the idea before this may be better though because it can raise exceptions when necessary. Exceptions are a good thing. We just need to find a good way to keep them from being thrown when the wrapper isn't needed

chibill, 2016-07-24:
Yeah.

nocarryr, 2016-08-18:
Closed by #46 
Thanks for the idea and for being patient!

closed 2016-08-18 by nocarryr