pyrtlsdr/pyrtlsdr #48: Error Code -1

mnager, 2016-08-28:
I am running Ubuntu and while I can import rtlsdr, when I try to initialize it, I get the following error:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/rtlsdr/rtlsdr.py", line 52, in __init__
    self.open(device_index, test_mode_enabled)
  File "/usr/local/lib/python2.7/dist-packages/rtlsdr/rtlsdr.py", line 68, in open
    % (result, device_index))
IOError: Error code -1 when opening SDR (device index = 0)

At another instance, I also got the same error, but with -3 instead of -1, in a manner identical to Issue #11.

Thanks,

Max/W2MAX

nocarryr, 2016-08-28:
As @roger- suggested in the issue you referenced, have you checked to make sure that `librtldsr` is able to communicate with your device? 
I would start with the built-in programs from it like `rtl_test`. Check out [their wiki](http://sdr.osmocom.org/trac/wiki/rtl-sdr) for command line arguments.

nocarryr, 2016-09-04:
@mnager have you had any luck with this?

mnager, 2016-09-05:
It's definitely an issue with rtl-sdr, not pyrtlsdr. After running 
rtl-test, I keep getting the same error:

usb_claim_interface error -6
Failed to open rtlsdr device #0

I used cmake with -DINSTALL_UDEV_RULES=ON and tried disabling 
the|||dvb_usb_rtl28xxu module, but to no avail(The module wasn't even 
loaded on one of the systems of tried.) So far, I have tried this on 
Ubuntu 16.04 and Raspbian. I did however, get qtcsdr working on a 
different pi, but that installs its own dependencies.|

roger-, 2016-09-05:
Try strace'ing rtl-test and see which call fails maybe? Have you tried running it as root?

mnager, 2016-09-11:
I have run it as root and that has made no difference. I can't tell what 
the issue is from strace'ing rtl-test. I can post the output here, if 
anyone thinks that will be useful.

At this point, I am just stumped and for the meantime, will just stick 
to using the rtl-sdr that comes with qtcsdr, so I can use pyrtlsdr.

Skaugent, 2016-11-09:
Hey,

same problem to me, I checked if rtl_test can use the dongle, so far no problems. But if I choose Python, then the same error again, but just if I try to specify the device index. 
I've to do this, because I will use 8 dongles in a cluster, it would be nice to hear if there is already a solution.

Maybe I've just done something wrong in the code, but however, it's possible to connect to different donfles like in GNURadio ? If so, how ?

Thanks for help and regards

Skaugent

Error-Message

Traceback (most recent call last):
  File "/media/R8000/Python_RTLSDR/PyRTLSDR.py", line 3, in <module>
    sdr = RtlSdr(3)
  File "/home/a8975/.local/lib/python2.7/site-packages/rtlsdr/rtlsdr.py", line 52, in __init__
    self.open(device_index, test_mode_enabled)
  File "/home/a8975/.local/lib/python2.7/site-packages/rtlsdr/rtlsdr.py", line 68, in open
    % (result, device_index))
IOError: Error code -1 when opening SDR (device index = 3)

nocarryr, 2016-11-09:
Does it fail for any of your non-zero indexed dongles? If you try 
```python
sdr = RtlSdr(1)
```
do you get the same error?

I would ask the same for index 0, but the two bits of code below would be doing the same thing under the hood:
```python
sdr = RtlSdr()
```
```python
sdr = RtlSdr(0)
```
I don't have more than one device, so unfortunately I can't duplicate the problem here.

Skaugent, 2016-11-09:
First, thanks for the quick answer, you maybe prevent me from a lot of work in GNURadio...
Ok, now surprise. It works, I tried with different dongle numbers, but it seems to work properly now. I will go on tomorrow and keep you up to date.
--> For now I was able to index up to 6 dongles

nocarryr, 2016-11-09:
Interesting.  I wonder if it has to do with the order of initialization. Did it work just now because you initialized them starting with 0 then going up to index 5?

Skaugent, 2016-11-10:
Yes I think so, I just made some initialization and here we have the outcome in the cmd...

Type "help", "copyright", "credits" or "license" for more information.
>>> from rtlsdr import *
>>> sdr = RtlSdr(0)
Found Rafael Micro R820T tuner
>>> sdr = RtlSdr()
usb_claim_interface error -6
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/a8975/.local/lib/python2.7/site-packages/rtlsdr/rtlsdr.py", line 52, in __init__
    self.open(device_index, test_mode_enabled)
  File "/home/a8975/.local/lib/python2.7/site-packages/rtlsdr/rtlsdr.py", line 68, in open
    % (result, device_index))
IOError: Error code -6 when opening SDR (device index = 0)
>>> sdr = RtlSdr(0)
usb_claim_interface error -6
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/a8975/.local/lib/python2.7/site-packages/rtlsdr/rtlsdr.py", line 52, in __init__
    self.open(device_index, test_mode_enabled)
  File "/home/a8975/.local/lib/python2.7/site-packages/rtlsdr/rtlsdr.py", line 68, in open
    % (result, device_index))
IOError: Error code -6 when opening SDR (device index = 0)
>>> sdr = RtlSdr(1)
Found Rafael Micro R820T tuner
>>> sdr = RtlSdr(0)
Found Rafael Micro R820T tuner
>>> sdr = RtlSdr(6)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/a8975/.local/lib/python2.7/site-packages/rtlsdr/rtlsdr.py", line 52, in __init__
    self.open(device_index, test_mode_enabled)
  File "/home/a8975/.local/lib/python2.7/site-packages/rtlsdr/rtlsdr.py", line 68, in open
    % (result, device_index))
IOError: Error code -1 when opening SDR (device index = 6)
>>> 

As you see, it's quite weird.  Another thing I was wondering about is, that I've now connected 6 Dongles to the computer with Serial number from 2-7 , but from index 5 up no SDR is recognized. 
I think the reason is, that the command RtlSdr(x) is just using the ID and not the Serial number... ?
Is it possible to use the serial number instead?

nocarryr, 2016-11-10:
Was that code you pasted in from a single session in python? It's hard to tell.  If you wrap individual blocks as [markdown code blocks](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) it would be a little easier to read.

The first two instances of `sdr` above would definitely fail if they're from the same session (without exiting the python interpreter) because calling
```python
sdr = RtlSdr()
```
is exactly the same as calling
```python
sdr = RtlSdr(0)
```
Also, if you're instantiating multiple sdr's, you'll need them to be assigned to either separate variables or placed in a container (`list`, `dict`, etc.) so that you have access to them and more importantly so they don't get garbage collected.

The garbage collection part is important because when a device gets gc'd, it will be closed by the library by [this method](https://github.com/roger-/pyrtlsdr/blob/master/rtlsdr/rtlsdr.py#L106-L107).

As far as the indexing is concerned, I don't know of a sure-fire method of knowing which device will be which other than using shell commands like `lsusb`.  The `device_index` is directly related to the underlying `librtlsdr` library.

nocarryr, 2016-11-11:
@Skaugent there is a way to get the `device_index` from its serial number:
https://github.com/librtlsdr/librtlsdr/blob/master/include/rtl-sdr.h#L58-L67

That function has not been wrapped yet though, so it may take a while to implement it.