pyrtlsdr/pyrtlsdr #49: IOError: Short read, requested 3000000 bytes, received 2999808

chibill, 2016-09-04:
When taking repeated reads of 3000000  bytes each it will eventually throw this error.

No matter the number of bytes requested the number of missing byte is the same.

I can take 728 reads just fine. (Each at a different center freq) 
Then on the 729 read_sample call it crashes.

nocarryr, 2016-09-04:
Have you checked your resource usage?  Memory-wise?  3 million samples seems like quite a lot for one call to `read_samples`.  I've never even tried reading more than 8192 in a single request.

chibill, 2016-09-04:
it works for 725 sample reads then that data is junked after ebing used. I will try calling GC and getting some memory info

nocarryr, 2016-09-04:
It could also be some sort of blocking I/O issue.  If possible, I would try reading smaller 'chunks', but more often.  If your application requires contiguous sampling, I think looking into the async methods would be ideal

chibill, 2016-09-04:
I most me some memory problem if I lower to just 1000 samples it works. But I am unsure of how good my Relative power levels I calculate will be.

roger-, 2016-09-05:
I second the idea of using async support (either callbacks or asyncio).

Picking the right buffer size has always been tricky, it seems to be OS dependent and it's also affected by whatever librtlsdr and libusb expects.

chibill, 2016-09-05:
Does that support changing the center freq after so many bytes? And does it also work properly with python 2.7

roger-, 2016-09-05:
I haven't tried changing the center freq, but it should work (some data will be corrupt due to settling,  etc.). 

Callbacks will work in 2.7.

nocarryr, 2016-09-05:
For the second question, the answer is "sort of".  You can use the `read_samples_async` method, but you won't be able to take advantage of asyncio, so the benefits aren't that great.

For the first question though, yes.  That's something you'll need to do in your code and also a good reason to read a smaller number of samples.

Read x number of samples and add them to a buffer or queue until you have enough.  Then process them, `set_center_freq` and do another run.

nocarryr, 2016-09-05:
Oh, and I figured a mention of async would get your attention @roger-  haha

Btw, I finally decided to get into actually using and understanding asyncio (in another project).  It's hard to understand at first, but definitely can `yield from` (pun intended) great results

roger-, 2016-09-05:
I `await` your thoughts on moving forward with async ;)

chibill, 2016-09-05:
I guess I will close this. (I got it working by using GC calls to force the GC to collect after each sample array is used  (After all the math if done that uses them))

closed 2016-09-05 by chibill

nocarryr, 2016-09-05:
Glad you found a solution.  Forcing garbage collection seems a bit hacky, but if it works, it works haha

roger-, 2016-09-05:
@chibill glad to hear it, guess this issue will require more investigation if it's potentially fixable from our end. For future reference could you post the fix you used?

chibill, 2016-09-05:
I just took less samples and ran gc.collect() when I finished using the samples before calling sdr.read_samples(Number of samples)