pyrtlsdr/pyrtlsdr #50: Whats the default bandwidth?

chibill, 2016-09-22:
Do you know what the bandwidth is considering we cant set it. (Yet)

Also as you can set the bandwidth with librtlsdr can this be impelemetned? (It works as I have used it using one of someone's fork. But you need the ip to data li rtlsdr)

nocarryr, 2016-09-22:
I don't have any experience with tuner bandwidth since I don't have a dongle that supports it, so I really don't know the answer to the first question.

I'm guessing that you're referring to [this fork](https://github.com/librtlsdr/librtlsdr) of librtlsdr though.  And I do remember your interest in [this branch](https://github.com/nocarryr/pyrtlsdr/tree/nocarryr-patch-2) from my fork of pyrtlsdr.

Are you mostly asking for the `rtlsdr_set_and_get_tuner_bandwidth` function to be implemented by pyrtlsdr?  (Just trying to make sure I'm understanding you correctly)

chibill, 2016-09-22:
Yes. And also the functions for it are in normal librtlsdr not not the latest binary so you have to use the forks binary. 

Most dongles support it. (Or else you could not use sdr# which sets the bandwidth.)

nocarryr, 2016-09-23:
It would seem that my RTL2832 dongle returns a bandwidth of 1200000 (1.2e6) at the default sample rate (1.204e6).

Since I can't set bandwidth on this tuner, it always returns the expected full-range bandwidth (following with the sample rate).
```python
>>> sdr.rs = 1.024e6
>>> sdr.get_bandwidth()
1200000L
>>> sdr.rs = 2.048e6
>>> sdr.get_bandwidth()
2050000L
```
I'll try and get a PR going for the implementation so others can test

chibill, 2016-09-23:
I have a RTL2832U which is basiclly the same thing and it supports it. (Make sure you have the librtlsdr with the function)

nocarryr, 2016-09-23:
Hmm.  I was only going off of [this commit message](https://github.com/librtlsdr/librtlsdr/commit/af20337c93ec04576011039ae1442ff83bfd0a87) that says it only works with the R820T.

Either way, changing the value does not seem to actually do anything for me.  I am positive I have the correct librtlsdr linked because the value from [this bit of code](https://github.com/nocarryr/pyrtlsdr/blob/nocarryr-patch-2/rtlsdr/librtlsdr.py#L131-L136) is `True` (the api function is available).

nocarryr, 2016-09-23:
Ok, now I'm feeling dumb. Just looked up the [dongle that I have](https://www.adafruit.com/product/1497) and yes, it's an RTL2832U with an R820T tuner. (Not an SDR expert)

It would seem that maybe I've got something wrong in the wrapper functions I wrote.  I'm not a ctypes expert either, so maybe @roger- could take a look at my code

nocarryr, 2016-11-11:
The merge of #51 should take care of this.  Unless there's more related discussion, I think we can close this now.  Will reopen otherwise.  Thanks!

closed 2016-11-11 by nocarryr