pyrtlsdr/pyrtlsdr #54: On Windows, 'rtlsdr_set_tuner_bandwidth' not found

Atheros1, 2016-12-19:
test.py fails on Windows:

    File "pyrtlsdr-master\rtlsdr\librtlsdr.py", line 139, in <module>
        f = librtlsdr.rtlsdr_set_tuner_bandwidth

It looks like this function was added in this commit
https://github.com/roger-/pyrtlsdr/commit/0687c7781425641be8f52916adbd89ac961aab12

I'm guessing that rtlsdr.dll just needs to be recompiled. I cannot find one that works that is already compiled. Does this mean that pyrtlsdr does not support Windows right now unless we can figure out how to compile rtlsdr ourselves?

nocarryr, 2016-12-24:
I don't know exactly which version the included dll file is to be honest, but most of the newer functionality for librtlsdr is being added in [this fork](https://github.com/librtlsdr/librtlsdr) of the original repository.

As for Windows builds, I haven't attempted since I only have one Windows machine (my son's YouTube watching machine).  I'll see if I can find any documentation for Windows compilation, but just in case it's something related to pyrtlsdr, can you please paste in the full traceback?  Does it show a return code?

Atheros1, 2016-12-25:
Oh I didn't know about that fork. Using that fixed it!

[image]

Beautiful!

closed 2016-12-25 by Atheros1

chibill, 2017-01-05:
@Atheros1 which windows one did you use. (I can't find one that will load.)

Atheros1, 2017-01-05:
I used the latest Windows release here:
https://github.com/librtlsdr/librtlsdr/releases
There is no 64 bit version so I had to download 32 bit Python and use that (it says "ImportError: Error loading librtlsdr. Make sure librtlsdr (and all of its dependencies) are in your path" otherwise. *rolls eyes*). 
I also tested that driver first by running rtl_test.exe first. It probably failed and it looks like I got libgcc_s_dw2-1.dll from somewhere and plopped that in the folder too. If you put all of those .dll files and everything in the pyrtlsdr folder then it can find them.

chibill, 2017-01-05:
Yeah. That lib is like hard to find... I had it on my old laptop apparently so I might try pulling it off it's hard drive. On my old laptop I had mingw installed and that's where it found that lib that is missing.

Atheros1, 2017-01-05:
I want to slap whoever invented this whole DLL dependency system. It's absurd.

chibill, 2017-01-05:
Yeah. Blame windows.

chibill, 2017-01-06:
I had to copy like 3 files from my old mingw install. Maybe we should compile a copy of librtlsdr that is staticly built so all the dlls are mashed into it.

nocarryr, 2017-01-06:
@Atheros1 @chibill I wish (sort of, but not really) that I could feel your pain, lol.  I guess I have been living in a *nix world for so long I wouldn't even know where to begin with building a lib in win. 
```
wget <librtlsdr-url>
tar xvzf <tar-filename>
cd librtlsdr/build
cmake ../
make
sudo make install
```
Can't imagine what you have to do in a non-shell environment, but if the links @Atheros1 added in #57 are helpful in this regard, it's a no-brainer to merge it (and also keep the links updated)

chibill, 2017-01-06:
Also on linux they have .deb builds for it. (makes it easier.)

nocarryr, 2017-01-06:
True, but they'll never be as up to date as a build off of master. Makes it
a real pain when you want to install GNU radio since it depends on outdated
built-in libs. We all have dependency issues I suppose, lol

chibill, 2017-01-06:
Well they have to be pretty close. (Same as all the windows builds.)