pyrtlsdr/pyrtlsdr #55: IOError: Error code -9 when setting center freq

Mitchell1314, 2016-12-24:
Hi,roger,
    I am using pyrtlsdr recently. I have a class get_samples:
     class get_samples(threading.Thread):
            global sdr
            def run(self):
                 global sdr

                 sdr.center_freq=104.1e6
                 sdr.sample_rate = 1.024e6
                 sdr.gain = 4
                 sdr.read_samples_async(cb, 1024 * 500) 
when I run it like:
          t1=get_samples()
          t1.run()
         sdr.set_center_freq(105e6)
But I got an error like:
         sdr.set_center_freq(105e6)
  File "C:\Python27\lib\site-packages\rtlsdr\rtlsdr.py", line 106, in set_center_freq
    % (result, freq))
IOError: Error code -9 when setting center freq. to 105000000 Hz

How can I fix it? waiting for your reply. Thank you!

nocarryr, 2016-12-24:
From the [libusb docs](http://libusb.sourceforge.net/api-1.0/group__misc.html) the error code is `LIBUSB_ERROR_PIPE`.  That isn't incredibly descriptive, but one thing that seems suspicious is concurrency in your program.

I can't tell because the formatting got a little jumbled, but are you sure that you are calling `cancel_read_async` and that the device is no longer reading samples **before** you try to change the center frequency?

Mitchell1314, 2017-03-02:
Thanks for your suggestion which has solved my problem perfectly.

closed 2017-09-27 by nocarryr