pyrtlsdr/pyrtlsdr #56: Error code -1 when reading 25600000 bytes

kehak, 2016-12-31:
I'm having this issue on Linux (tried on Fedora 25 and Ubuntu 16.04) when acquiring some samples:

```
Found Rafael Micro R820T tuner
Traceback (most recent call last):
  File "./sdrfft.py", line 18, in <module>
    samples = (sdr.read_samples(sr*5)) # in secs
  File "/usr/lib/python2.7/site-packages/rtlsdr/rtlsdr.py", line 336, in read_samples
    raw_data = self.read_bytes(num_bytes)
  File "/usr/lib/python2.7/site-packages/rtlsdr/rtlsdr.py", line 320, in read_bytes
    % (result, num_bytes))
**IOError: Error code -1 when reading 25600000 bytes**
```
The dvb-usb module is blacklisted, the udev rules are set properly, rtl_sdr works great as a limited user and can acquire for a long time with no drop outs.

rtl_test output:
```
$ rtl_test -s 2.56e6
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
Sampling at 2560000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
```

Here's a snippet of my code:
```

secs = 5
fft_length = 4096
win = signal.blackmanharris(fft_length)

sdr = RtlSdr()
sdr.DEFAULT_ASYNC_BUF_NUMBER = 16 # tried that to solve the pb, it didn't
sr = 2.56e6  # Hz
sdr.sample_rate = sr
sdr.center_freq = 1400e6     # Hz
sdr.gain = 48

samples = (sdr.read_samples(sr*secs)) # in secs
```
It will work if secs<4, any value above will cause this issue. On Windows I can acquire 20s with no problem.

Any idea?

nocarryr, 2017-01-02:
I would recommend reading smaller sets of samples and concatenating them together.  In your snippet, you're making one call to `read_samples(sr*secs)`.  With the given values, the number of samples for `read_samples()` would be 12,800,000 and the actual number of bytes read (for `sdr.read_bytes()`) is twice that.

In my experience, I try to keep the number of samples read in a single call to no more than 8,192, but your mileage may vary.

Is it possible in your use case to do something like:
```python
N = int(sr * secs)  # must be integer, would need to be checked for remainder (N += 1 if necessary) 
read_size = 4192
num_read = 0 # number of samples read

samples = None
while num_read <= N:
    sweep = sdr.read_samples(read_size)
    if samples is None:
        samples = sweep
    else:
        samples = np.concatenate((samples, sweep))
    num_read += read_size
if samples.size > N:
    samples = samples[:N]  # I think that slice expression is correct, would have to verify
```
That's assuming you're using numpy, a pure python version would be a bit different (and probably easier to read)

If your application requires contiguous samples (no gaps in time between reading), then you would want to look at `sdr.read_samples_async()`.  An example for its use can be found [here](https://github.com/roger-/pyrtlsdr/blob/master/tests/utils.py#L135-L150)

kehak, 2017-01-03:
Thank you nocarryr, both workarounds worked.

I used your first idea as this function just measures the gain and NF of a receiving chain, so I don't need the samples to be contiguous.
For larger acquisitions, I usually call rtl_sdr, dump the samples to a file and post-process it with Python/Numpy, so using `sdr.read_samples_async()` should be a much cleaner approach.

I'm not sure why reading a larger number of samples doesn't work anymore on Linux with pyrtlsdr...
I'll let Roger get back to me if he wants more information to investigate before closing the issue.

nocarryr, 2017-01-03:
Good to hear you got things going.  I imagine there could be a limit imposed somewhere in the low-level side.  Either in the array allocation done in [`sdr.read_bytes()`](https://github.com/roger-/pyrtlsdr/blob/master/rtlsdr/rtlsdr.py#L346-L349) or somewhere in librtlsdr?  Possibly even something in libusb. Since I'm not very well-versed in the low-level side....

Paging @roger-

closed 2017-09-27 by nocarryr

yuvadm, 2018-05-26:
This seems like a pretty weird limitation not being able to read a 'large' amount of samples.

nocarryr, 2018-07-19:
Under any circumstance I can imagine yes, you would be losing information between calls to `sdr.read_samples()`.  I'm not sure what the exact limit is, but a single method call isn't the best idea if you're wanting to collect samples for more than a few seconds.

The best approach for this is IMO would be to use asyncio and `sdr.read_samples_async()`.  Internally, it pulls a continuous stream from the device that you read from periodically, so you can avoid losing samples.

See example here:
https://nocarryr.github.io/pyrtlsdr/rtlsdraio.html

This of course would require Python >= 3.5

hemanti0503, 2018-07-20:
Thank you! I'm going to look into this.

janispritzkau, 2019-05-21:
You could try rounding the number of samples to the nearest multiple of 512 or 1024.

`sdr.read_samples((sample_rate * seconds + 511) // 512 * 512)`

It seems to work fine for me.