pyrtlsdr/pyrtlsdr #59: Expose the ir interface (For the ir sensor on most Rtl-Sdrs) from librtlsdr.

chibill, 2017-01-09:
I will be working on a PR for this. (So I actually help some)

nocarryr, 2017-01-09:
Great! I've been thinking about adding that, but haven't been able to dedicate much time lately.

Feel free to create a PR even if it's not complete (just add [WIP] to the title).  I or someone else may be able to help review as it progresses

chibill, 2017-01-09:
Okay. I am working on finding other stuff that we could expose to while I am digging around  in librtlsdr. (Like GPIO control for people with rtl-sdrs from rtl-sdr.com  that has a bias-T controled fro mthe gpios)

chibill, 2017-01-09:
https://github.com/librtlsdr/librtlsdr/blob/master/src/librtlsdr.c#L2408 is the function that deals with reading the IR sensor.

nocarryr, 2017-01-09:
To get the public API calls, you'll want to look at [include/rtl-sdr.h](https://github.com/librtlsdr/librtlsdr/blob/master/include/rtl-sdr.h#L439)

You'll see a lot of similarity between that and [librtlsdr.py](https://github.com/roger-/pyrtlsdr/blob/master/rtlsdr/librtlsdr.py)

chibill, 2017-01-09:
Okay. Did not really know that was even there. 👍

chibill, 2017-01-09:
I added the function into librtlsdr the interface in rtlsdr.py might give me alittle trouble but. (I am not very good at ctypes yet.) Not sure how to make the c_uint8 pointer thing that is the buf for the result.