pyrtlsdr/pyrtlsdr #62: Could not set sample rat to 30kHz

p1mike, 2017-03-24:
Hi Roger,
could use rtf_fm with 30kHz sample rate. Pls give some advise.
Thank you

Invalid sample rate: 30000 Hz
Traceback (most recent call last):
  File "./run_rtl_sdr.py", line 9, in <module>
    sdr.sample_rate = 0x7530  # Hz
  File "build/bdist.linux-armv7l/egg/rtlsdr/rtlsdr.py", line 171, in set_sample_rate
IOError: Error code -22 when setting sample rate to 30000 Hz

itdaniher, 2017-03-27:
@p1mike 30kHz is well below the ~200kHz minimum sample rate. rtl_fm internally sets a sampling rate in the 250kHz or so range and decimates to the specified bandwidth. 

This is not an issue with pyrtlsdr.

nocarryr, 2017-03-27:
@itdaniher Thanks for responding to this.  Your help is much appreciated!

@p1mike Apologies for the delay, but yes, as @itdaniher stated, that error is coming from outside of pyrtlsdr.

The wrapper is designed to throw an `IOError` whenever an issue is detected from the underlying `librtlsdr` (C library) or `libusb`.

There are quite a few methods for sampling outside of the device's tuning range, but I am far from an RF expert

closed 2017-03-27 by nocarryr