pyrtlsdr/pyrtlsdr #63: Does not work on 64-bit python.

chibill, 2017-04-04:
I can't get pyrtlsdr to work on 64-bit python. I am 99% sure its not a pyrtlsdr problem but a problem with librtlsdr not being compiled for 64-bit. But I am still opening an issue here in case someone also has the problem.

nocarryr, 2017-04-04:
Have you tried building librtlsdr from source? I know there isn't much documentation out there on building for windows, but I just stumbled on this:
https://github.com/librtlsdr/librtlsdr/tree/master/win32-qtcreator

The readme seems to have at least a bit of info to get things rolling.
I would also recommend opening an issue on [librtlsdr](https://github.com/librtlsdr/librtlsdr) to see if they have a way of including x64 builds in their releases.  I can't tell exactly how they're doing the builds at first glance, but it seems they've got it going for quite a few platforms.

chibill, 2017-04-05:
I made in issue on there end but they have not responded yet.

chibill, 2017-04-07:
Well this sort of sucks... They have not responded to anyone in over a week.... And I don't want to downgrade my python to 32 bit on my 64 bit machine... :(

So currently I can't use this lib :(

chibill, 2017-04-09:
Okay. One of the devs says they will try building it for 64-bit tomorrow... 

Apparently they did not build it for 64-bit because most people only give out 32 bit.....

https://github.com/librtlsdr/librtlsdr/issues/51

closed 2017-04-09 by chibill