pyrtlsdr/pyrtlsdr #64: syntax error

fanisg, 2017-04-29:
I have Installed it with
pip install -U rtlsdr_scanner
running it
python -m rtlsdr_scanner
gives
```
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.6/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.6/runpy.py", line 153, in _get_module_details
    code = loader.get_code(mod_name)
  File "<frozen importlib._bootstrap_external>", line 781, in get_code
  File "<frozen importlib._bootstrap_external>", line 741, in source_to_code
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
  File "/usr/lib/python3.6/site-packages/rtlsdr_scanner/__main__.py", line 39
    print 'Import error: {}'.format(error)
                           ^
SyntaxError: invalid syntax
```
and does not run. Is there a problem with pip install?

nocarryr, 2017-04-29:
This is Python 3 compatibility issue with the `print` statement used in the `rtlsdr_scanner` project, not this library (`pyrtlsdr`).
If you need to use Python 3, I would suggest opening an issue here:
https://github.com/EarToEarOak/RTLSDR-Scanner

Or, if you don't absolutely need to use Python 3, you should be able to install Python 2.7, repeat the installation as before and as far as I know, be up and running.

nocarryr, 2017-09-27:
@fanisg were you able to get this working?  I'm going to close this issue for now unless there's anything left unresolved

closed 2017-09-27 by nocarryr