pyrtlsdr/pyrtlsdr #65: Issue with Mac Book Pro

etcj, 2017-07-03:
I have got a new Mac Book Pro and pyrtlsdr fails to work. After trying your previous suggestion:
import ctypes
from ctypes.util import find_library
file_name = find_library('rtlsdr')
dll = ctypes.cdll.LoadLibrary(file_name)
print (dll)
print (dll.rtlsdr_get_device_count), the following resulted:
<CDLL '/usr/local/lib/librtlsdr.dylib', handle 100600ab0 at 0x1022a1780>
<_FuncPtr object at 0x102239a70>, which seems to be the expected result.
However, when I try to instantiate the class, I get: error code -1, when opening SDR (device_index=0).
This also happen with other device numbers.
Any hint of what can be wrong? Thank you in advance.

etcj, 2017-07-03:
Please note that it runs without problems under Matlab, Simulink or rtl_test. The above error only results when running pyrtlsdr under Python 3.6 and Python 2.7.

nocarryr, 2017-07-06:
I have no direct experience with compiling or installing the shared `librtlsdr` library, but it seems that your system is able to find it.  Did you install it using homebrew or some other method?  The only information I can go off of is from #24 

Here's one thing that may or may not help in determining whether the wrapper is able to access the library:
```python
import rtlsdr
print(rtlsdr.librtlsdr)
# >>>    <CDLL '/usr/local/lib/librtlsdr.dylib', handle............>
print(rtlsdr.librtlsdr.rtlsdr_get_device_count())
# >>>    1 (or however many devices you have attached to your system)
```

Also, if there is any other information given in the error on instantiation, can you please paste it in a [code block](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)

etcj, 2017-07-07:
After executing the above commands:
1 - rtlsdr.lbrtlsdr resulted in: <CDLL '/usr/local/lib/librtlsdr.dylib', handle 1007477a0 at 0x1022a2080>
2 - rtlsdr.librtlsdr.rtlsdr_get_device_count(): resulted in 0.

I also checked again rtl_test, with positive results.

Please also note that I have no problems with the Matlab installation, that uses the same library, Gqrx and Cocoa.

Yes. The library has been installed with brew. However, the Matlab installation and Gqrx worked before the installation with brew.

Could it be some sort of limitation imposed by Mac OS Sierra? I am aware that it prevents execution of libraries that are not certified by Apple.

Your library (pyrtlsdr) has been installed with pip (native Python 2.7), pip3 (Python 3.5 and 3.6) and using your setup file. Nothing different happened.

When the RtlSdr class is instantiated, the following message results:
Traceback (most recent call last):
  File "ex1.py", line 3, in <module>
    sdr = RtlSdr()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rtlsdr/rtlsdr.py", line 52, in __init__
    self.open(device_index, test_mode_enabled)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rtlsdr/rtlsdr.py", line 68, in open
    % (result, device_index))
OSError: Error code -1 when opening SDR (device index = 0)

Thank you very much for you help so far.

nocarryr, 2017-07-07:
> Could it be some sort of limitation imposed by Mac OS Sierra? I am aware that it prevents execution of libraries that are not certified by Apple.

This may or may not be the case since other external applications are working properly (using the shared/public library.  I wouldn't rule it out though.

If `librtlsdr.dylib` were completely unavailable, this would make sense.  But it appears to be locating and accessing it:
`<CDLL '/usr/local/lib/librtlsdr.dylib', handle 1007477a0 at 0x1022a2080>`
instead of raising an exception and failing completely on import.

It could still be something related to this though.  Possibly permissions with the device or with `libusb` itself.  Since we aren't currently testing with OSX or Python3.6, it's difficult to say for sure.  I'll find an iMac somewhere that I can use as a test bed.

Also, have you run the last test you mentioned using Python2 (or anything less than 3.6)?

etcj, 2017-07-07:
Dear Roger,

Thank you very much for your help.

The test with Python 2.7.10 (the natively installed by Apple) results in error:
Traceback (most recent call last):
  File "test1.py", line 1, in <module>
    import rtlsdr
  File "build/bdist.macosx-10.12-intel/egg/rtlsdr/__init__.py", line 56, in <module>
  File "build/bdist.macosx-10.12-intel/egg/rtlsdr/librtlsdr.py", line 139, in <module>
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ctypes/__init__.py", line 378, in __getattr__
    func = self.__getitem__(name)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ctypes/__init__.py", line 383, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: dlsym(0x7fc733c1fb20, rtlsdr_set_tuner_bandwidth): symbol not found

I also tried with RtlTcp, running the dongle external on a Raspberry pi. The code was as follows:
from pylab import *
from rtlsdr import *
client = RtlSdrTcpClient(hostname='10.0.1.67', port=12345)
client.center_freq = 115e6 # the dongle is connected to a RF signal generator
samples = client.read_samples(256*1024)
# use matplotlib to estimate and plot the PSD
psd(samples, NFFT=1024, Fs=sdr.sample_rate/1e6, Fc=sdr.center_freq/1e6)
xlabel('Frequency (MHz)')
ylabel('Relative power (dB)')
show()

The error msg:
Traceback (most recent call last):
  File "ex1.py", line 5, in <module>
    client.center_freq = 92.5e6
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rtlsdr/rtlsdrtcp/client.py", line 86, in set_center_freq
    self._communicate_descriptor_set('fc', value)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rtlsdr/rtlsdrtcp/client.py", line 80, in _communicate_descriptor_set
    return self._communicate(msg)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rtlsdr/rtlsdrtcp/client.py", line 56, in _communicate
    s = self._build_socket()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rtlsdr/rtlsdrtcp/client.py", line 42, in _build_socket
    s.connect((self.hostname, self.port))
ConnectionRefusedError: [Errno 61] Connection refused

I also tested on another Mac, with similar results. I am starting to believe that this has something to do with the internals of Mac OS Sierra.

Best Regards,

Eutiquio

nocarryr, 2017-07-07:
I think I see the issue here:
```
AttributeError: dlsym(0x7fc733c1fb20, rtlsdr_set_tuner_bandwidth): symbol not found
```
The `rtlsdr_set_tuner_bandwidth` function has only been a part of librtlsdr for about a year and may not included in the version available from homebrew (based off of the original [librtlsdr](https://github.com/steve-m/librtlsdr) repo.  That was added in [this commit](https://github.com/librtlsdr/librtlsdr/commit/af20337c93ec04576011039ae1442ff83bfd0a87) from the [currently active fork](https://github.com/librtlsdr/librtlsdr).

Is it possible for you to compile from source?  If not, I can add a case to handle the missing function and generate a new release.

etcj, 2017-07-07:
Thank you very much. That is very kind of you to offer a new release, which I accept. Compiling from sources within Mac environment is not always an easy task, although I have done that before.

nocarryr, 2017-07-07:
No problem.  I just pushed the release through PyPI.  You should be able to update by
```bash
pip install -U pyrtlsdr
```
(add `sudo` if needed)

It should install version `0.2.5`.  Let me know if that solves things for you.

etcj, 2017-07-07:
Please note that I am using your library with Ubuntu, at another computer, and also with Raspberry Pi. Congratulations for the beautiful piece of work you have done so far. I wish I could contribute a bit in the future. My background is more on the RF and electromagnetic side, although I am able to program in C, C++, C# and Python, despite my limited talent with programming.

nocarryr, 2017-07-07:
Honestly, my knowledge in those areas is still quite limited (even after quite a bit of Khan Academy).  So any input you have would be greatly appreciated.

etcj, 2017-07-07:
I am afraid that this still did not solve the problem.
Under Python 2.7.10:
from rtlsdr import RtlSdr
sdr = RtlSdr()

Error:
Traceback (most recent call last):
  File "ex1.py", line 3, in <module>
    sdr = RtlSdr()
  File "/Library/Python/2.7/site-packages/rtlsdr/rtlsdr.py", line 58, in __init__
    self.open(device_index, test_mode_enabled)
  File "/Library/Python/2.7/site-packages/rtlsdr/rtlsdr.py", line 74, in open
    % (result, device_index))
IOError: Error code -1 when opening SDR (device index = 0

Under Python 3.6:
Traceback (most recent call last):
  File "ex1.py", line 3, in <module>
    sdr = RtlSdr()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rtlsdr/rtlsdr.py", line 58, in __init__
    self.open(device_index, test_mode_enabled)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rtlsdr/rtlsdr.py", line 74, in open
    % (result, device_index))
OSError: Error code -1 when opening SDR (device index = 0)

nocarryr, 2017-07-07:
Interesting.  I think I may be out of ideas until I can sit in front of a Mac to play with. The only thing I can think of is that the device is still in use or the handle is still open from a prior attempt.  I can only suggest unplugging and re-attaching the dongle in between tests at this point (which I'm sure you've already done)

etcj, 2017-07-07:
Yes. I did. In fact, I also rebooted the computer. Sorry about that.

etcj, 2017-07-07:
When within Matlab and with the same dongle:
hn = sdrinfo()

hn = 

                RadioName: 'Generic RTL2832U OEM'
             RadioAddress: '0'
              RadioIsOpen: 0
                TunerName: 'R820T'
             Manufacturer: 'Realtek'
                  Product: 'RTL2838UHIDIR'
               GainValues: [29×1 double]
      RTLCrystalFrequency: 28800000
    TunerCrystalFrequency: 28800000
             SamplingMode: 'Quadrature'
             OffsetTuning: 'Disabled'
Which shows it is working.

The same dongle from a Terminal:
mbproEC:~ ec$ rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...