pyrtlsdr/pyrtlsdr #69: Multiple RTLSDR dongles using in one PC

Xie-Tian, 2017-08-14:
Hi, 
I want to use to two Rafael Micro R820T tuner dongles to take data from two sources.
The center frequencies, sample_rates, gains and every setting are the same for two dongles.
I was wondering that how can I separate two dongles in my script of Python and to take data from one source that I want, not another, at one time.
I tried to change my RTL-SDR files following the enhancement

> [WIP] Support addressing devices by serial number #66

However, I don't know how to use it...
Thanks!

nocarryr, 2017-08-14:
If you don't need to address them by their serial numbers, you can use the normal released version of pyrtlsdr.  You would instantiate them by index (starting with 0):
```python
from rtlsdr import RtlSdr

sdr1 = RtlSdr(device_index=0)
sdr2 = RtlSdr(device_index=1)

# Maybe just check to make sure they are working properly
sdr1.center_freq = 100e6
sdr2.center_freq = 200e6
print(sdr1.center_freq)
print(sdr2.center_freq)
```
Otherwise, with the changes in #66 you can retrieve the detected serial addresses as a list with the staticmethod `get_device_serial_addresses` [here](https://github.com/nocarryr/pyrtlsdr/blob/e34108efd2ad0c751e02948be64622b531758d71/rtlsdr/rtlsdr.py#L66) then use the `serial_number` argument when creating an instance:
```python
serials = RtlSdr.get_device_serial_addresses()
print(serials)

# Using "serials[0]", but this could be the string itself
sdr1 = RtlSdr(serial_number=serials[0])
```

Xie-Tian, 2017-08-14:
That's cool!!! Thanks a lot!!!!

closed 2017-08-14 by Xie-Tian