pyrtlsdr/pyrtlsdr #71: what kind of data we can receive ?

Toutoupou, 2017-08-18:
Hi ! I wouldlike to know what kind of data the RTLSDR2832 can receive I try the script python with PSD and i use this fonction : 20*log(samples_read) to have data in dB but the problem I don't know if this data is in dbm or not

nocarryr, 2017-09-27:
I apologize for leaving this question out there for so long, but I was hoping someone more knowledgeable than myself in the RF/DSP field would come across it.

It can be a bit difficult to reason about the result of a PSD in terms of dB, specifically the reference point. (dB with reference to ....)

It depends on the signal being fed into the device (antenna, cabling, etc), both in impedance and overall gain.  There are many other factors so it would be difficult to provide any concrete answer to this.  I would recommend reading through:
* Some topics on [stack-exchange](https://dsp.stackexchange.com):
  * https://dsp.stackexchange.com/questions/32038/rtl-sdr-units-textrmdbm
  * https://dsp.stackexchange.com/questions/tagged/sdr
* https://dspguru.com/
* https://www.dsprelated.com/

(Just to name a few)

It may be helpful to find a known reference point (a signal with known characteristics that you can receive), that you can calibrate your device with.

ordonezt, 2020-08-29:
The data is normalized to the maximum value readable by the ADC, in case of an 8bits ADC that value is (2 ^ 8) - 1 = 255.
But this data has an offset of 127 ADC counts (the middle of ADC voltage reference) that needs to be substracted.
You can see it in this lines of code of [rtlsdr.py](https://github.com/roger-/pyrtlsdr/blob/master/rtlsdr/rtlsdr.py#L495-L505) file.
```
    def packed_bytes_to_iq(self, bytes):
        """Unpack a sequence of bytes to a sequence of normalized complex numbers
        This is called automatically by :meth:`read_samples`.
        Returns:
            The unpacked iq values as either a :class:`list` or
            :class:`numpy.ndarray` (if available).
        """
        if has_numpy:
            # use NumPy array
            data = np.ctypeslib.as_array(bytes)
            iq = data.astype(np.float64).view(np.complex128)
            iq /= 127.5
            iq -= (1 + 1j)
        else:
            # use normal list
            iq = [complex(i/(255/2) - 1, q/(255/2) - 1) for i, q in izip(bytes[::2], bytes[1::2])]

        return iq
```
So with this `20 * log (samples_read)` you can get the difference of `samples_read` with respect to 1 in dB

But i have a question, according to the [RLT2832U datasheet](https://www.kechuang.org/r/30360) the chip has a 7-bit ADC (page Nº2)
Shouldn't we divide the data by (2 ^ 7 - 1) = 127 and then substract 127 / 2 = 63.5?

Also in page Nº5 we can see information about the voltage reference of the ADC (I don't understand wich of 3,3V or 1,2V is the correct value)
So with this value we can get the voltage signal, multiplying the samples by the reference voltage.

jabowery, 2024-08-27:
> ...
>             iq = [complex(i/(255/2) - 1, q/(255/2) - 1) for i, q in` izip`(bytes[::2], bytes[1::2])]
> 

I'm confused by the reference to `izip` because I've been trying to figure out if I can use `pyrtlsdr` to read in a file created by the `rtl_sdr` command line, which has real and imaginary bytes interleaved.  That's when I ran across the `packed_bytes_to_iq` method only to discover that I can't import `izip` from `itertools` because it is only supported in `python**2.7**`!  This is really strange because it would appear from this that `pyltrsdr` isn't compatible with `python3`.

nocarryr, 2024-08-29:
@jabowery  De-interleaving method uses numpy if available and falls back to using the `zip` built-in function.  This wasn't available in earlier Python versions (possibly even before 2.7) so `izip` is actually a reference to `zip` in pretty much all cases:
https://github.com/pyrtlsdr/pyrtlsdr/blob/0450698ff50a45c149b2fe6d31b51055723a8099/rtlsdr/rtlsdr.py#L27-L28

So the line in question is actually:

```python
iq = [complex(i/(255/2) - 1, q/(255/2) - 1) for i, q in zip(bytes[::2], bytes[1::2])]
```

You could also just pull it directly in as a numpy array which is arguably more efficient:
https://github.com/pyrtlsdr/pyrtlsdr/blob/0450698ff50a45c149b2fe6d31b51055723a8099/rtlsdr/rtlsdr.py#L653-L656

Or if you'd rather keep your dtype's more compact, you could substitute `np.float64` for `np.float32` and `np.complex128` for `np.complex64`.