pyrtlsdr/pyrtlsdr #72: [R82XX] PLL not locked! Patch Included

tomdean1, 2017-09-02:
The '[R82XX] PLL not locked!' message on initialization is caused by the order of commands in rtlsdr.py
Changing the order eliminates the warning message

> diff -c rtlsdr.py.orig rtlsdr.py
*** rtlsdr.py.orig      Sat Sep  2 01:50:36 2017
--- rtlsdr.py   Sat Sep  2 01:50:42 2017
***************
*** 93,100 ****
          self.valid_gains_db = [val/10 for val in self.gain_values]
  
          # set default state
-         self.set_sample_rate(self.DEFAULT_RS)
          self.set_center_freq(self.DEFAULT_FC)
          self.set_gain(self.DEFAULT_GAIN)
  
      def close(self):
--- 93,100 ----
          self.valid_gains_db = [val/10 for val in self.gain_values]
  
          # set default state
          self.set_center_freq(self.DEFAULT_FC)
+         self.set_sample_rate(self.DEFAULT_RS)
          self.set_gain(self.DEFAULT_GAIN)
  
      def close(self):

nocarryr, 2017-09-02:
I'm trying to understand the diff output you pasted.  Are you suggesting that the order should become:

1. `set_center_freq`
2. `set_sample_rate`
3. `set_gain`

I agree that the warning message is a bit annoying, but will have to test the results later on.

Please feel free to submit a PR (looks like it could be your first pull request, so yay!).  A change like this one would be pretty simple and I can't imagine it would cause any side effects.

Clicking on the edit button in [rtlsdr.py](https://github.com/roger-/pyrtlsdr/blob/master/rtlsdr/rtlsdr.py) will allow to make the changes and submit them.

https://help.github.com/articles/editing-files-in-another-user-s-repository/

chibill, 2017-09-25:
This message is also given by normal RTLSDR binary files. (Like rtl_power) So not sure its that big of a problem.

tomdean1, 2017-09-25:
It adds un-necessary noise to the process.  If it is wrong, it is wrong!

Tom Dean

chibill, 2017-09-25:
But does that also mean the person who originally created the driver is using it wrong?

nocarryr, 2017-09-27:
It seems as though `rtl_power` gives the same warning just as a coincidence.  (Setting gain, sampling mode, bias tee and tuning offset **before** setting the sample rate):
https://github.com/librtlsdr/librtlsdr/blob/master/src/rtl_power.c#L930-L953

At least that's how I'm interpreting things.  I don't see any issues with changing the order of things on initialization within `pyrtlsdr` unless it has some adverse effect I'm unaware of.

Thoughts anyone?

nocarryr, 2017-09-28:
@tomdean1 Looks like you've already made the changes in [your fork](https://github.com/tomdean1/pyrtlsdr/tree/patch-1).

Please feel free to submit a pull request