pyrtlsdr/pyrtlsdr #73: How to export IQ data (question)

natiya, 2017-10-30:
Is there any way to record and export IQ data as a file .wav file, for example?

nocarryr, 2017-10-30:
If you have scipy installed, you can use [scipy.io.wavfile.write](https://docs.scipy.org/doc/scipy/reference/generated/scipy.io.wavfile.write.html#scipy.io.wavfile.write).

The sample rate should match `sdr.sample_rate`.  The "in-phase" (real) is typically placed in the left channel and the "quadrature" (imag) part in the right channel.

In order to format the array into the expected shape, you could do something like:
```python
from scipy.io import wavfile

N = 4096 # number of samples
samples = sdr.read_samples(N)
wav_samples = np.zeros((N, 2), dtype=np.float32)

wav_samples[...,0] = samples.real
wav_samples[...,1] = samples.imag

wavefile.write('out.wav', int(sdr.sample_rate), wav_samples)
```

Forgive me if I messed up the slicing syntax and I know there are better methods of reshaping.

Also, since the values returned from `RtlSdr.read_samples` are normalized to -1 to 1, the wav file would end up being 32-bit which isn't an exact replica of the data returned directly from the device (8-bit unsigned).  It would take a bit more work to get that (you would have to use `RtlSdr.read_bytes` and handle the buffer pointers yourself).

nocarryr, 2017-10-30:
If you just need to save the data as a file however, you could just use numpy's [save function](https://docs.scipy.org/doc/numpy/reference/generated/numpy.save.html#numpy.save).  No data manipulation would be needed, but you'd want to make sure to note the sample rate (and other parameters).

```python
import numpy as np
from rtlsdr import RtlSdr

sdr = RtlSdr()

N = 4096
samples = sdr.read_samples(N)
np.save('samples.npy', samples)

# then to open later
samples = np.load('samples.npy')
```

natiya, 2017-10-31:
@nocarryr thanks a lot. The second option, which looks simpler, saves the data as an .npy file, but I would need a .wav file because I need to load it in a signal generator.

I tried to run the first option but it doesn't work. I think there is a problem with `from scipy.io import wavfile`
Thank you!

nocarryr, 2017-10-31:
@natiya what version of scipy are you using?  Maybe try 
```python
import scipy.io.wavile
scipy.io.wavefile.write()
```

Alternatively, you could use the [wave module](https://docs.python.org/3.6/library/wave.html) built into the stdlib, but you'll need to do a bit more work with formatting the data beforehand.

natiya, 2017-10-31:
@nocarryr I'm using version 3.6 because my Python is that version too. So I tried the following script:

```
import numpy as np
import scipy.io.wavile
from rtlsdr import RtlSdr

sdr = RtlSdr()

N = 4096 # number of samples
samples = sdr.read_samples(N)
wav_samples = np.zeros((N, 2), dtype=np.float32)

wav_samples[...,0] = samples.real
wav_samples[...,1] = samples.imag

scipy.io.wavefile.write('out.wav', int(sdr.sample_rate), wav_samples)
```

Maybe I wrote something wrong?