pyrtlsdr/pyrtlsdr #74: error when getting the device serial number

natiya, 2017-11-16:
when I do:

`serial_numbers = RtlSdr.get_device_serial_addresses()`

I get this error:

> 'Error code %d when reading USB strings (device %d)' % (r, device_index)
> OSError: Error code -12 when reading USB strings (device 0)

This is the method:

```
def get_device_serial_addresses():
        def get_serial(device_index):
            bfr = (c_ubyte * 256)()
            r = librtlsdr.rtlsdr_get_device_usb_strings(device_index, None, None, bfr)
            if r != 0:
                raise IOError(
                    'Error code %d when reading USB strings (device %d)' % (r, device_index)
                )
            return ''.join((chr(b) for b in bfr if b > 0))

        num_devices = librtlsdr.rtlsdr_get_device_count()
        return [get_serial(i) for i in range(num_devices)]
```
and:

```
# int rtlsdr_get_device_usb_strings(uint32_t index, char *manufact,
#                                   char *product, char *serial)
f = librtlsdr.rtlsdr_get_device_usb_strings
f.restype, f.argtypes = c_int, [c_uint,
                                POINTER(c_ubyte),
                                POINTER(c_ubyte),
                                POINTER(c_ubyte)]
```
What do I need to change to get it to work? Thanks!

nocarryr, 2018-06-05:
According to the [libusb docs](http://libusb.sourceforge.net/api-1.0/group__libusb__misc.html), error code -12 is `LIBUSB_ERROR_NOT_SUPPORTED`.  I'm not sure why the device wouldn't support that function though since it's fairly general.

Are you able to communicate with the device at all?

lockemoor, 2018-10-13:
i guess this is like a year overdue, but i had a similar problem and it solved when i installed the osmocom driver