pyrtlsdr/pyrtlsdr #76: MacOS librtlsdr

jrl290, 2018-01-20:
`brew install librtlsdr`

Installs .a and .dylib files. Seems pyrtlsdr only references .dll and .so files

nocarryr, 2018-01-20:
The library search mechanism used ( [`ctypes.find_library`](https://docs.python.org/3.6/library/ctypes.html#ctypes.util.find_library) ) does not depend on filename extensions being supplied properly since it handles the platform-specific logic internally.

In `pyrtlsdr` that function call can be found on the last line [here](https://github.com/roger-/pyrtlsdr/blob/v0.2.7/rtlsdr/librtlsdr.py#L21-L25):
```python
def load_librtlsdr():
    driver_files = ['rtlsdr.dll', 'librtlsdr.so']
    driver_files += ['..//rtlsdr.dll', '..//librtlsdr.so']
    driver_files += ['rtlsdr//rtlsdr.dll', 'rtlsdr//librtlsdr.so']
    driver_files += [find_library('rtlsdr'), find_library('librtlsdr')]
```
Since you're installing `librtlsdr` through homebrew, are you doing the same for `pyrtlsdr`?
`brew install pyrtlsdr` ?

Are you sure that `librtlsdr` has been installed successfully? The `rtl_test` program is a good indicator.

I'm not sure how homebrew handles python packages that aren't specifically built (brewed?) for it.  Does it use its own package index or does it search for candidates on [PyPI](https://pypi.python.org/pypi)?  It seems from reading the [docs](https://docs.brew.sh/Homebrew-and-Python.html) that the preferred method is to use `pip install`, but I could be wrong.

jrl290, 2018-01-20:
So here's my error:
```
Traceback (most recent call last):
  File "/Users/jrl290/Documents/NNFM/Generate.py", line 4, in <module>
    from rtlsdr import *
  File "/Library/Python/2.7/site-packages/rtlsdr/__init__.py", line 56, in <module>
    from .librtlsdr import librtlsdr
  File "/Library/Python/2.7/site-packages/rtlsdr/librtlsdr.py", line 41, in <module>
    librtlsdr = load_librtlsdr()
  File "/Library/Python/2.7/site-packages/rtlsdr/librtlsdr.py", line 36, in load_librtlsdr
    raise ImportError('Error loading librtlsdr. Make sure librtlsdr '\
ImportError: Error loading librtlsdr. Make sure librtlsdr (and all of its dependencies) are in your path
```

But here's what I get when I run find_library:
```
>>> from ctypes.util import find_library
>>> find_library('rtlsdr')
'librtlsdr.dylib'
>>> find_library('librtlsdr')
'librtlsdr.dylib'
```

Any ideas?

jrl290, 2018-01-20:
Seems brew installs version 0.5.3 whereas the latest version is 0.7

Let me see if I can work with that

nocarryr, 2018-01-20:
Try this:

```python
from ctypes.util import find_library
from ctypes import CDLL

libname = find_library('librtlsdr')
lib = CDLL(libname)
```

`find_library` only returns the name.  `CDLL` is the mechanism that loads it.

jrl290, 2018-01-20:
Installed 0.7.0: No change

```
>>> from ctypes.util import find_library
>>> from ctypes import CDLL
>>> libname = find_library('librtlsdr')
>>> CDLL(libname)
<CDLL 'librtlsdr.dylib', handle 7fc3b340ace0 at 105eee9d0>
```

jrl290, 2018-01-20:
I installed pyrtlsdr with pip, but the line numbers in the error seem to be off from the github sources. Could that be a clue?

nocarryr, 2018-01-20:
The line numbers reflect changes from #75.  That change should only affect linux users and since you were previously using an older version with the same issue, I don't think it's related.

The version you have installed should match this:
https://github.com/roger-/pyrtlsdr/blob/v0.2.7/rtlsdr/librtlsdr.py

If you can edit the source file, maybe we can get some more information:
"/Library/Python/2.7/site-packages/rtlsdr/librtlsdr.py"

On line 30 change this:
```python
        try:
            dll = CDLL(driver)
            break
        except:
            pass
```
to this:
```python
        try:
            dll = CDLL(driver)
            break
        except:
            if 'dylib' in driver:
                raise
```
The `raise` keyword should throw the exception instead of silently ignoring it.  Then maybe there will be some useful error message

jrl290, 2018-01-20:
```
Traceback (most recent call last):
  File "/Users/jrl290/Documents/NNFM/Generate.py", line 4, in <module>
    from rtlsdr import *
  File "/Library/Python/2.7/site-packages/rtlsdr/__init__.py", line 56, in <module>
    from .librtlsdr import librtlsdr
  File "/Library/Python/2.7/site-packages/rtlsdr/librtlsdr.py", line 42, in <module>
    librtlsdr = load_librtlsdr()
  File "/Library/Python/2.7/site-packages/rtlsdr/librtlsdr.py", line 31, in load_librtlsdr
    dll = CDLL(driver)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ctypes/__init__.py", line 365, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(/usr/local/lib/librtlsdr.dylib, 6): no suitable image found.  Did find:
	file system sandbox blocked open() of '/usr/local/lib/librtlsdr.dylib'
```

jrl290, 2018-01-20:
Ok, so it seems my Python editing application, PythonRunner, was stuck in a sandbox. It runs fine from the command line.

My apologies. Thank you for your help

nocarryr, 2018-01-21:
No problem. Glad you got it working

closed 2018-01-21 by nocarryr