pyrtlsdr/pyrtlsdr #77: discontinuities in read_samples

niamiot, 2018-01-22:
Hi,

I'm facing a strange behaviour when I use the read_samples method: A discontinuity appears at the beginning of the data. 
Here's my setup:
- My sdr dongle ( [this reference](http://www.nooelec.com/store/sdr/sdr-receivers/nesdr-smart-sdr.html) ) is directly connected to a signal generator.
generator central freq is 434 MHz output = -20 dBm.
- I use the following piece of code to get my samples and plot the result : 

```python
import rtlsdr
import matplotlib.pyplot as plt
sdr=rtlsdr.RtlSdr()
sdr.set_gain(0)
sdr.set_center_freq(434e6)
sdr.set_bandwidth(10e3)
X=sdr.read_samples(4096*2)
X=sdr.read_samples(4096*2) # twice to ensure gain setup has been taken into account
plt.plot(X)
```

and here's an example of the output I get:

[image]

Does someone has any clue on what's going on ?

nocarryr, 2018-01-22:
Looks like out of phase information at the beginning. That would make sense since you're plotting complex (quadrature) samples.  What happens if you plot `X.real` and `X.imag` separately?

niamiot, 2018-01-29:
Hi,

No I doesn't seems to be a phase information. I just update my code to differentiate both real and imag part and I've got the same issue

```python
import rtlsdr
import matplotlib.pyplot as plt
sdr=rtlsdr.RtlSdr()
sdr.set_gain(0)
sdr.set_center_freq(434e6)
sdr.set_bandwidth(10e3)
X=sdr.read_samples(4096*2)
X=sdr.read_samples(4096*2) # twice to ensure gain setup has been taken into account
plt.subplot(211)
plt.plot(X.real)
plt.subplot(212)
plt.plot(X.imag)
```

[image]

The issue appears always at the same sample (1802) whatever the bandwidth or frequency. For a different sample rate the discontinuity is shifted (e.g. 1834 @2.88e6 ).

nocarryr, 2018-06-05:
@niamiot Have you tried adding a brief pause before reading samples?  (After the gain and center frequency is set)

I've seen that as an issue in quite a few places.  Adding a `time.sleep(.1)` to allow the device to settle is not uncommon.

```python
sdr.set_gain(0)
sdr.set_center_freq(434e6)
sdr.set_bandwidth(10e3)
time.sleep(.1)
X = sdr.read_samples(8192)
```

Since it's always at the exact same sample though, it's hard to say for sure that's the issue.  1802 samples at 2.88e6 = 0.000626 seconds, seems very precise.  Still worth a try though.