pyrtlsdr/pyrtlsdr #79: Return relative power

Wiki61, 2018-02-21:
Is it possible to return the relative power from the plot like in your example?
https://github.com/roger-/pyrtlsdr#plotting-the-psd-with-matplotlib

For example freq = 95.5e6 and have it return -30

roger-, 2018-02-21:
You can use the same PSD function to look up the power at whatever frequency you like. Alternatively you can use a narrowband filter and then take the mean squared value of the output to find the power.

Wiki61, 2018-02-21:
Would you happen to have an example of either of those options?

roger-, 2018-02-21:
Something like this:

f, P = scipy.signal.welch(x, Fs)

ind = np.abs(1e6 - f).argmin()

power = P[ind] # convert to log if needed

mconlow, 2020-11-19:
I think this question is how to make the chart in the example code come back with a more familiar dbm scale with values < 0. Mine for example had some single digit positive numbers with the example code. When I changed the psd line to this:

`psd(samples, NFFT=1024, Fs=sdr.sample_rate/1e6, Fc=sdr.center_freq/1e6, scale_by_freq=False)`

the scale moved down into a more familiar all negative scale. I'm not sure exactly what the scale_by_freq is doing, maybe someone with more experience in this can say

nocarryr, 2020-11-19:
Per the [matplotlib docs](https://matplotlib.org/3.1.1/api/_as_gen/matplotlib.pyplot.psd.html), `scale_by_freq` normalizes the values to units of `Hz^-1` or `V/Fs` if `True`, resulting in "Power Spectral Density". If `False`, the results are `V^2` (magnitude squared) resulting in simply "Power". Source code for the conversion can be found [here](https://github.com/matplotlib/matplotlib/blob/e125ffbae5662bf215ff258148e52bcae999921b/lib/matplotlib/mlab.py#L415)

Those two options depend on your application and are both "correct", even with positive results in decibels. This would be expected given that decibels are by definition relative.

If you're looking for results in actual [dBm](https://en.m.wikipedia.org/wiki/DBm) , the reference point would need to be determined for your specific device.