pyrtlsdr/pyrtlsdr #84: How to reduce the center frequency spike?

JSzut, 2018-06-14:
Is there a way to reduce the center frequency spike using the library itself?

nocarryr, 2018-06-14:
Is your question related to spectral analysis?  There shouldn't necessarily be a "spike" at the center frequency for the raw IQ samples read from the device.

You may be seeing that in an FFT or PSD, depending on the methods used.  Can you provide a code example?

itdaniher, 2018-06-14:
You can see some excellent discussion [here](https://wiki.analog.com/resources/eval/user-guides/ad-fmcomms1-ebz/iq_correction) - the spur at 0Hz relative to your tune frequency is a fact of the mismatch between the in-phase and quadrature (I&Q) branches of the (rtl)sdr - besides modeling and subtracting the offset, there's not much you can do.

nocarryr, 2018-06-15:
Great answer itdaniher I always thought that was an issue with the 0 Hz frequency bin of the Fourier transfer. Never knew it had to do with the LO down mix on the device itself, so thanks again for taking me to school, lol

michelp, 2018-06-15:
Thanks @itdaniher ! great info

JSzut, 2018-06-15:
Thank you for your involvment guys. Special thanks to @itdaniher for his answer.

dave-pi, 2019-11-12:
While IQ imbalance is caused by mismatches in the two paths, RX DC offset is mainly caused by 3 factors:

- LO signal self mixing due to finite isolation between LO ports
- LO signal (self or interfering) re-radiation from the antenna
- Interfering signal and circuit generated second-order nonlinearities

I suggest the following article which explains these fenomena in more detail http://venividiwiki.ee.virginia.edu/mediawiki/images/9/93/DCR_Raman.pdf.
Most commercial grade integrated solutions, include DACs in the Rx front end path, to perform a calibration (to null the DC offset presented at the ADC).
You can in software calculate the average of the signal and subtract it, but it's a non-ideal solution.

RobCoding, 2022-12-11:
> While IQ imbalance is caused by mismatches in the two paths, RX DC offset is mainly caused by 3 factors:
> 
>     * LO signal self mixing due to finite isolation between mixer ports
> 
>     * LO signal (self or interfering) re-radiation from the antenna
> 
>     * Interfering signal and circuit generated second-order nonlinearities
> 
> 
> I suggest the following article which explains these fenomena in more detail http://venividiwiki.ee.virginia.edu/mediawiki/images/9/93/DCR_Raman.pdf. 

That website no longer exists, try: https://web.archive.org/web/20170808211704/http://venividiwiki.ee.virginia.edu/mediawiki/images/9/93/DCR_Raman.pdf