pyrtlsdr/pyrtlsdr #85: Listen to two radio simultaneously?

patrickcgray, 2018-07-04:
Has anyone used the ability to handle multiple devices to successfully listen to two rtl_sdr devices simultaneously? I want to do some comparisons of a signal from antennas at different locations at the same time and I'm not sure how to do this. Do I need to get them to run in separate threads? or multiple processes? Any examples of getting this to work?

michelp, 2018-07-04:
I don't know the specific answer to your question, and maybe you already know this, but if you want to correlate the same signals in two receivers, not just power received, the receivers will need to have a synchronized frequency source, for example see https://www.rtl-sdr.com/building-a-quad-rtl-sdr-receiver-for-radio-astronomy/.

Another use people have had is to have multiple receivers scanning different frequencies, for this you don't need combined TXCO, but there are cpu/io challenges see: http://lukeberndt.com/2016/using-multiple-rtl-sdr-to-capture-a-trunking-system/

patrickcgray, 2018-07-09:
I'm really looking for a rough relative comparison of signal strengths from two directional antenna so I can get a rough position estimate (or at least heading compared to the two antenna) of a nearby radio signal. 

I was able to do this pretty effectively in python 2.7 by starting two processes and then having them wait to start sampling until a predefined time. This typically gets them to finish a 2 second sampling run within 10 ms of each other. I'd like to get closer to ~1 ms and in python 3 there seem to be some good options like Barrier but I haven't quite found the right solution in python 2.7.

wgaylord, 2018-08-22:
Might be able to use processing and open both devices. One on each process.

patrickcgray, 2018-08-26:
I ended up getting this to work relatively well. Not quite as in sync as I wanted but I did as @wgaylord suggested and used processing. The two SDRs, run on a raspberry pi, usually finish with ~10ms. The following code got it to work:

```
def run_sdr(sdr, name, return_dict, num_samples, start_time=time()):
    
    print(name)
    now_start = time()
    # this function waits to begin until the specified start_time and basically runs psd()
    power_db, psd_freq = sample_spectrum(sdr, num_samples, start_time=start_time)
    now_end = time()
    print(name, now_start, now_end)
    return_dict[name] = [power_db, psd_freq]
    
def run_sdrs(sdr1, sdr1_name, sdr2, sdr2_name, num_samples):
    manager = Manager()
    output_dict = manager.dict()

    start_time = time() + 1
    left = Process(target = run_sdr, args=(sdr1, sdr1_name, output_dict, num_samples, start_time))
    right = Process(target = run_sdr, args=(sdr2, sdr2_name, output_dict, num_samples, start_time))
    left.start()
    right.start()
    #print('Start time:', start_time)
    left.join()
    right.join()
    return(output_dict)
```

closed 2018-08-26 by patrickcgray