pyrtlsdr/pyrtlsdr #91: 3.7 needs s/async def __aiter__/def __aiter__/

itdaniher, 2018-12-06:
Small issue with Py3.7 - one line change.

L134 @ rtlsdraio.py should be `    def __aiter__(self):`

nocarryr, 2018-12-06:
IIRC, there's a bit of a backwards compatibility issue with that...

In Py3.4 I think an awaitable had to be returned, then Py3.5+ expected an async iterator but would raise PendingDeprecationWarning.  In Py3.7 I'm guessing it becomes an error?
https://docs.python.org/3.6/reference/datamodel.html#asynchronous-iterators
https://docs.python.org/3.7/reference/datamodel.html#asynchronous-iterators

itdaniher, 2018-12-06:
yeah, I read that this breaks 3.4, haven't tested on 3.5/3.6.

nocarryr, 2018-12-07:
Just realized that we are only supporting aio in Py3.5+ since the `async def` throws a SyntaxError when parsing the class body (that's documented).
So I don't see anything blocking that change. Do you want to do a PR?

closed 2018-12-15 by nocarryr