pyrtlsdr/pyrtlsdr #94: Always similar graph output? Center freq. spike?

KeyLo99, 2019-01-15:
I'm trying to implement FFT and plot samples that I get from RTL-SDR device. But I'm always getting the similar output and a spike at the center of the graph even though I change the center frequency.
The code that I'm using is test.py which is provided at this repository. Is this problem related to my device?

[image]

[image]

nocarryr, 2019-01-15:
@KeyLo99 This is common and expected behavior. Check out issue #84 for some discussion on the topic

roger-, 2019-01-15:
There are algorithms you can implement to estimate the error and then compensate. Not sure if Python implementations exist anywhere.

itdaniher, 2019-01-15:
https://gist.github.com/itdaniher/7ac96d5e6f477b3d0e4bb03f6df466dd

I don't have source for this excerpt, but had it saved in my "interesting bits of SDR code" folder. Might be of use here.

see also: https://wiki.analog.com/resources/eval/user-guides/ad-fmcomms1-ebz/iq_correction

nocarryr, 2019-01-15:
Wow, looks very adaptive.  I wonder how the `DcTimeConst` value (`5e-5f`) was determined

itdaniher, 2019-01-15:
@nocarryr I wouldn't fret about the college edu - learned less relevant stuff from my ECE professors than I did from ie) https://dspguide.com/ / wikipedia.

I would like to think it works with uncorrelated / whitenoise, but I'm not willing to speak authoritatively as of yet. Fundamentally you're just trying to solve for the... 2x2 offset/rotation matrix - also expressible as phase/amplitude error. It varies at least a little bit with regards to time, temperature, frequency, etc (all the things ADCs can vary with, and then some!), which is maybe / probably why some folks use iterative approaches.

two pdfs to call out
* https://www.faculty.ece.vt.edu/swe/argus/iqbal.pdf (simple / short, from the ADI page)
* https://vtechworks.lib.vt.edu/bitstream/handle/10919/78202/Inti_D_T_2017.pdf (sweet looking thesis I'm gonna have to check out this evening)

nocarryr, 2019-01-15:
Thanks! I'll add those to my library give them a read

KeyLo99, 2019-01-15:
Thanks for the awesome gist and explanatory sources. But I'm wondering, do you have the same "IQ balancer" implementation in C? Later I might change my mind from Python to C. If you don't have it or don't know any sources, I will rewrite :)

nocarryr, 2019-01-15:
I imagine it would be fairly straightforward to create a wrapper for airspy so that you can use it in python.

There are a few approaches
1. [ctypes](https://docs.python.org/3.7/library/ctypes.html) which is in python's standard library.  A good example would be this project itself
2. [CFFI](https://cffi.readthedocs.io/en/latest/) even though I haven't used it much, its designed to minimize the amount of work involved with translating header code
3. [Cython](https://cython.readthedocs.io/en/latest/) my favorite these days, I don't use it much for wrapping C/C++ libs, but for the performance gains (generating C from typed python)

itdaniher, 2019-01-16:
@KeyLo99 I don't have any closed-form implementations of IQ balancing in C handy, definitely easier to do with numpy IMHO.

KeyLo99, 2019-01-16:
@itdaniher I have to implement IQ Correction algorithm in C because I'm using librtlsdr [C]. Another reason is I didn't find any good C implementation except the non-documented and complex ones. So I want to write a clean and simple code that handles IQ Correction and other stuff.
Would you tell me what algorithm you used in that Python gist? Or leave some links, PDFs, etc.

itdaniher, 2019-01-17:
@KeyLo99 found it the source at https://www.reddit.com/r/RTLSDR/comments/uje3q/sample_python_code_and_results_for_iq_correction/

I don't think "clean and simple" + "DSP in C" is a semantically valid construct.... [liquid-dsp](https://github.com/jgaeddert/liquid-dsp/) might save you some pain, but that's not likely to result in an implementation that could be upstreamed into librtlsdr.

roger-, 2019-01-18:
Knew that code looked familiar 😂

nocarryr, 2019-01-18:
Hahaha, I didn't notice that at first @roger-

KeyLo99, 2019-01-18:
Okay, I kinda solved the problem with playing with the antenna and cables, changing the receiver position. Central peak disappeared but it looks like I/Q data that I get from device still needs some fixing and balance. I will try to apply any filter/algorithm to my raw I/Q and FFT [out] data to fix this imbalance issue. I'm open to your suggestions :)

KeyLo99, 2019-01-19:
Yeah looks like so...
I'm gonna make this code public for SDR researchers/enthusiasts when I'm done fixing issues about efficiency. I hope you guys will tell me the points that I don't get and I should fix :)

itdaniher, 2019-01-19:
Please do!

KeyLo99, 2019-01-27:
Alright, there you go: https://github.com/KeyLo99/rtl_map

Waiting for feedbacks :)

closed 2020-09-10 by orhun