pyrtlsdr/pyrtlsdr #95: Strange error running pyrtlsdr example

wangshujun1, 2019-02-28:
>>> from rtlsdr import RtlSdr

sdr = RtlSdr()

# configure device
sdr.sample_rate = 2.048e6  # Hz
sdr.center_freq = 70e6     # Hz
sdr.freq_correction = 60   # PPM
sdr.gain = 'auto'

print(sdr.read_samples(512))

>>> %Run read.py
  File "/home/pi/Documents/read.py", line 1
    Python 3.5.3 (default, Jan 19 2017, 14:11:04)
             ^
SyntaxError: invalid syntax

I just copied the code in readme.md and successfully installed setup.py. Is this something wrong with librtlsdr?
I'm very new to python and pyrtlsdr, hope someone can help

wangshujun1, 2019-02-28:
btw i test.py and demo_waterfall work well

wangshujun1, 2019-03-03:
no one??

roger-, 2019-03-03:
Can you post the entire code?

Are you missing an import at the top?

wangshujun1, 2019-03-03:
> Run the code from a terminal using 'python read.py' instead.

It did not work and shown the same error

nocarryr, 2019-03-03:
It seems likely that there's a copy-paste issue.  Can you paste the contents of `read.py` as a code block?

Start the code block with three backticks:
\```

Then paste your code on the next line and end with another triple backtick line \```

https://help.github.com/en/articles/basic-writing-and-formatting-syntax#quoting-code

nocarryr, 2019-03-03:
It looks like this is what you originally pasted:

```python
>>> from rtlsdr import RtlSdr

sdr = RtlSdr()

# configure device
sdr.sample_rate = 2.048e6  # Hz
sdr.center_freq = 70e6     # Hz
sdr.freq_correction = 60   # PPM
sdr.gain = 'auto'

print(sdr.read_samples(512))
```

If that's the case, the first line is what's causing problems.  Just remove the `>>> `  and any spaces on that line so it looks like:
```python
from rtlsdr import RtlSdr
```

wangshujun1, 2019-03-03:
> It looks like this is what you originally pasted:
>
> ```python
>
> sdr = RtlSdr()
>
> # configure device
> sdr.sample_rate = 2.048e6  # Hz
> sdr.center_freq = 70e6     # Hz
> sdr.freq_correction = 60   # PPM [...]

still did not work

wangshujun1, 2019-03-03:
ohhh  I fixed it!
I found that error is ''Python 3.5.3 (default, Jan 19 2017, 14:11:04) 
[GCC 6.3.0 20170124] on linux
Type "copyright", "credits" or "license()" for more information.'' which created by Python itself. It worked after I deleted this. Thanks for the help!

closed 2019-03-03 by wangshujun1