pyrtlsdr/pyrtlsdr #96: Access violation reading

montegry, 2019-03-19:
Windows 7 (32). Python 3.7. Trying to read (1024*256) samples through timerEvent every 1sec. and draw it on plot. First time he read samples with no issues, second time I get "Access violation reading".

`def timer_event(self):
        print("timer event start iter number:", self.iter)
        try:
            samples = self.rtl.read_samples(1024*4*16)
        except Exception as e:
            print(e)
        print("samples readed iter:", self.iter)
        self.power, power_freq, power_t = specgram(samples, mode='magnitude')
        #print(get_magnitude_spectrum[0][:20], '\n', get_magnitude_freqs[:20], '\n', get_magnitude_t[:20])
        self.magnitude, _, _ = specgram(samples, mode='psd')
        self.phase, _, _ = specgram(samples, mode='phase')
        #print("power 0 =", len(self.power[0]), '\n power freq=', len(power_freq), '\n power t =', len(power_t))
        self.axe1.clear()
        self.axe2.clear()
        self.axe3.clear()
        self.axe1.plot(range(len(self.magnitude[0])), self.magnitude[0])
        self.axe2.plot(range(len(self.power[0])), self.power[0])
        self.axe3.plot(range(len(self.phase[0])), self.phase[0])
        self.fig.canvas.draw()
        print("timer event end iter:", self.iter, '\n')
        self.iter += 1`

Log:
timer event start iter number: 0
samples readed iter: 0
timer event end iter: 0 

timer event start iter number: 1
exception: access violation reading 0x00000008
samples readed iter: 1

Process finished with exit code 255

montegry, 2019-03-20:
I "solve" this porblem by using TCP server and client.

nocarryr, 2019-08-13:
It seems like your timer event might be firing before the first iteration finished.

You may be able to solve this by guarding the `read_samples()` call with a lock:

```python
import threading

READ_LOCK = threading.Lock()

def timer_event():
    with READ_LOCK:
        samples = self.sdr.read_samples(1024*4*16)
    # do stuff with samples
```