pyrtlsdr/pyrtlsdr #99: How can I want to get the value at some freq?

weathon, 2019-07-30:


nocarryr, 2019-08-10:
Can you please clarify what it is you're asking?

weathon, 2019-08-13:
@nocarryr  Like I am doing a project the measure the signal strength at a particular frequency, so I want to get the value of it.

nocarryr, 2019-08-13:
For basic tuning and capturing, you can follow [this example](https://nocarryr.github.io/pyrtlsdr/Overview.html#examples) in the docs. You'll want to use a center frequency that is near (but not equal to) the frequency you're looking for.

After capturing samples, you can use something along the lines of of [scipy.signal.welch](https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.welch.html#scipy.signal.welch) to get the "estimated power spectral density" across the collected samples.

weathon, 2019-08-13:
OK, thanks. But why I should use a center frequency that is near **(but not equal to)**?

weathon, 2019-08-13:
@nocarryr

nocarryr, 2019-08-14:
There are potential issues with a "spike" at the center frequency.  Check out #94 for details

weathon, 2019-08-14:
OK, thanks

dustingrady, 2020-01-25:
Weathon, were you ever able to sort this out? I'm also interested in being able to determine the power/ amplitude at a given frequency.

dustingrady, 2020-01-26:
@nocarryr Thank you for answering this question. 

I am using pyrtlsdr in an attempt to read signal strength at a given frequency; It sounds like scipy.signal.welch might be what I should be using as well. 

The welch function takes a few arguments, one of which is an 'array like' argument.

In the examples on pyrtlsdr it has a line
`    samples = sdr.read_samples(256*1024)
`
Is this the array that should be fed to welch?

The `Power spectrum of x` returned from welch looks to be an array with very small values (ranging from x.xx e-1 to x.xx e-5). I'm guessing I will need to find a way to match these values with with the range of frequencies collected.

Sorry if this is a big question, just trying to get my head around this.

nocarryr, 2020-01-26:
@dustingrady yes, the `samples` array returned from
```python
samples = sdr.read_samples(256*1024)
```
is the correct input argument to either of the two methods (`scipy.signal.welch` or `matplotlib.mlab.psd`)

Both of them compute [Estimated Power Spectral Density](https://ccrma.stanford.edu/~jos/mdft/Power_Spectral_Density_Estimation.html) and the returned `Pxx` (power array) contains the magnitude squared result (very small values are to be expected).

After calculating the PSD, one typically would prefer to scale the result to [decibels](https://ccrma.stanford.edu/~jos/mdft/Decibels.html), and this is where it gets a bit complicated.

Since a decibel represents an intensity **relative** to some other (reference) value, one immediately wonders *"what's reference point for these values I just captured through a USB dongle with a random antenna and an arbitrary gain setting?"*

I don't have a concrete answer to that question honestly, you could calculate the reference point using the antenna impedance, the peak-to-peak voltages, and lots of other engineering stuff that I wish I knew more of.

The simple approach would be to just *not* use a reference point:
```python
import numpy as np
from scipy import signal

num_samples = 256*1024
samples = sdr.read_samples(num_samples)
freqs, Pxx = signal.welch(samples, fs=sdr.sample_rate)

# Use the 'power' formula for dB (10*log10(X))
# The 'np.abs(Pxx)' is there because 'Pxx' is complex-valued
amplitude_dB = 10 * np.log10(np.abs(Pxx))
```

I hope that helps.  I could expand more, but many blogs and books can and have been filled with this topic

nocarryr, 2020-01-26:
@dustingrady after re-reading your question, I realized I didn't actually answer it, lol

The return values from both `welch` and `psd` are two separate [numpy arrays](https://numpy.org/doc/1.17/reference/arrays.ndarray.html) with:

* `Pxx` containing the magnitude-squared results and
* `freqs`  containing the frequencies corresponding to the elements in `Pxx`

If you're wanting to combine them into a single array, I would recommend keeping them as numpy arrays so you can perform further math operations on them later.

You could use:
### [structured data types](https://numpy.org/doc/1.17/glossary.html#term-structured-data-type)

```python
# dtype with 'freqs' (float) and 'values' (complex)
data_type_complex = np.dtype([
    ('freqs', np.float64),
    ('values', np.complex128),
])

# dtype with 'freqs' (float) and 'values' (float)
data_type_real = np.dtype([
    ('freqs', np.float64),
    ('values', np.float64),
])

freqs, Pxx = signal.welch(samples, fs=sdr.fs)

# Create an empty array the same size as the result using the dtype above
psd_array = np.zeros(freqs.size, dtype=data_type_complex)

# Put the results in the new array
psd_array['freqs'] = freqs
psd_array['values'] = Pxx
```
Then you can access the fields using index or field name lookups

### [record arrays](https://numpy.org/doc/1.17/reference/routines.array-creation.html#creating-record-arrays-numpy-rec)

```python
freqs, Pxx = signal.welch(samples, fs=sdr.fs)
psd_array = np.core.records.fromarrays([freqs, Pxx], names='freqs,values')
```
**Note:** I haven't tested the code above and haven't really used record arrays very much, so I can't give much more advice on the subject

dustingrady, 2020-01-27:
@nocarryr Thank you so much for the responses; Very helpful!

You clarified the part that was most confusing for me; The small results in the (Pxx) power array. Is converting these values to decibel using the power formula a necessity, or is it manly for readability?

I am using your example on the pyrtlsdr page, and by adding the frequency with the corresponding power value to a dictionary, I can see that higher amplitudes approach and even cross into positive values [example](https://i.imgur.com/bsIeIQ2.png). However, when I use welch instead of psd, the frequencies returned [seem off](https://i.imgur.com/af6TZXj.png).  

Thank you again for your quick response and for creating this library, much appreciated :)

nocarryr, 2020-01-27:
You're correct on the last part. `matplotlib.mlab.psd` has a convenience argument for center frequency. Under the hood though, it just shifts the frequency array instead of leaving it centered around zero. 
If you don't supply that argument or are using `scipy.signal.welch` (or some other method), you can do the same thing by:
```python
freqs, Pxx = psd(samples, sdr.fs)

# Shift frequencies by the center frequency during sample stage
freqs += sdr.fc
```