pyrtlsdr/pyrtlsdr #101: TCP doesn't work

weathon, 2019-08-08:
code
```python
from rtlsdr import *

client = RtlSdrTcpClient(hostname='192.168.0.24', port=14423)
client.center_freq = 2e6
data = client.read_samples()
```

Output
```bash
wayne@wayne-X502CA:~/Ubuntu$ python3 tcp-test.py 
Traceback (most recent call last):
  File "tcp-test.py", line 4, in <module>
    client.center_freq = 2e6
  File "/home/wayne/.local/lib/python3.6/site-packages/rtlsdr/rtlsdrtcp/client.py", line 86, in set_center_freq
    self._communicate_descriptor_set('fc', value)
  File "/home/wayne/.local/lib/python3.6/site-packages/rtlsdr/rtlsdrtcp/client.py", line 80, in _communicate_descriptor_set
    return self._communicate(msg)
  File "/home/wayne/.local/lib/python3.6/site-packages/rtlsdr/rtlsdrtcp/client.py", line 57, in _communicate
    resp = tx_message.send_message(s)
  File "/home/wayne/.local/lib/python3.6/site-packages/rtlsdr/rtlsdrtcp/base.py", line 304, in send_message
    return self.get_response(sock)
  File "/home/wayne/.local/lib/python3.6/site-packages/rtlsdr/rtlsdrtcp/base.py", line 193, in get_response
    return cls.from_remote(sock)
  File "/home/wayne/.local/lib/python3.6/site-packages/rtlsdr/rtlsdrtcp/base.py", line 235, in from_remote
    kwargs = json.loads(header)
  File "/usr/lib/python3.6/json/__init__.py", line 354, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.6/json/decoder.py", line 339, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.6/json/decoder.py", line 357, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)```

nocarryr, 2019-08-09:
Were there any errors in the server process?

There are a couple of things to try that might help in finding the problem:

Try calling some other methods to see if their responses throw an error:
```python
from rtlsdr import *

client = RtlSdrTcpClient(hostname='192.168.0.24', port=14423)
current_fc = client.get_center_freq()
print(current_fc)
resp = client.set_center_freq(2e6)
print(resp)
# data = client.read_samples()
```

Or better yet, do this to retrieve the raw data coming back from the server:
```python
import json
from rtlsdr import *

client = RtlSdrTcpClient(hostname='192.168.0.24', port=14423)
try:
    current_fc = client.get_center_freq()
    print(current_fc)
    resp = client.set_center_freq(2e6)
    print(resp)
    # data = client.read_samples()
except json.JSONDecodeError as exc:
    print('<FROM SERVER>')
    print(exc.doc)
    print('</FROM SERVER>')
    raise
```

weathon, 2019-08-13:
@nocarryr I am not using this library to do the server, I am using rtl_tcp to do the server on my Android phone

nocarryr, 2019-08-13:
Unfortunately, the TCP module in pyrtldsr is not compatible with `rtl_tcp`.
I will add a note in the docs to avoid confusion in the future

closed 2019-08-13 by nocarryr

weathon, 2019-08-17:
OK, thanks