pyrtlsdr/pyrtlsdr #103: Crashing when using multiprocessing/pyqt5

pbutler, 2019-08-13:
When trying to use PyQT5 and multiprocessing Process, I am getting the following error and crashes.  Not  entirely sure what's causing it but looks like it might be loading the ctypes stuff in such a way to cause this.  I suspect this only happens in the presence of PyQt5 because it causes something in the objc library to get marked as dirty.

```
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
```

A simple example of non-working code:
```python
from PyQt5 import QtWidgets
import multiprocessing as mp
from rtlsdr import RtlSdr

class TestWindow(QtWidgets.QMainWindow):
    def __init__(self):
        super().__init__()
        self.p = mp.Process(target=rtltest)
        self.p.start()

def rtltest(self):
     rtl = RtlSdr()

if __name__ == "__main__":
    app = QtWidgets.QApplication([])
    application = TestWindow()
    application.show()
    app.exec()
```

pbutler, 2019-08-14:
Just realized it might not be obvious but this is on OS X

nocarryr, 2019-08-14:
Have you tried moving the import statement into the target function?

```python
def rtltest(self):
    from rtlsdr import RtlSdr
    rtl = RtlSdr()
```

pbutler, 2019-08-14:
Same results, same error message.