pyrtlsdr/pyrtlsdr #105: Getting access violations when doing some actions.

wgaylord, 2019-08-19:
Seems something is a bit odd. Using 
Python 3.7.3 (v3.7.3:ef4ec6ed12, Mar 25 2019, 22:22:05) [MSC v.1916 64 bit (AMD64)] on win32

Setting a bad sample rate 0.9e6 will error then trying to do any other action on the sdr object throws access violations. 

Same goes for bad values for the freq_calibration such as 0. 

Basically anytime you get an error code the next action and all after will cause an access violation error.

Code to test

```
from rtlsdr import RtlSdr
sdr = RtlSdr()

sdr.sample_rate = 0.9e6 # Errors as it should.
sdr.sample_rate = 2.048e6 #Errors with access violation

#Reset for next error

sdr = RtlSdr()
sdr.freq_correction = 0 #Weird that it errors
sdr.freq_correction = 1 #Errors with access violation
```

No idea why this is happening.

nocarryr, 2019-08-19:
When any error is detected from librtlsdr (the C library), the device is closed to prevent further issues:

```python
    def set_sample_rate(self, rate):

        rate = int(rate)

        result = librtlsdr.rtlsdr_set_sample_rate(self.dev_p, rate)
        if result < 0:
            self.close()           # <--------------
            raise IOError('Error code %d when setting sample rate to %d Hz'\
                          % (result, rate))

        return
```
https://github.com/roger-/pyrtlsdr/blob/190d3ac4308dcf71b0ee3cc791a856852aa96038/rtlsdr/rtlsdr.py#L261-L271

I'm assuming that snippet was from an interactive session.  If it were a script, the interpreter would exit after the first error forcing you to re-launch.

In interactive mode, it's likely that the file handles, pointers, etc in librtlsdr and libusb are still floating around and when you try to reuse them, bad things happen.  That's my guess anyway.

Best method would be to exit the interpreter and re-open

wgaylord, 2019-08-19:
I wonder how I never ran into this stuff happening before. Okay.

closed 2019-08-19 by wgaylord

nocarryr, 2019-08-19:
I'm not entirely sure, but it looks like this may be related:

https://github.com/steve-m/librtlsdr/commit/be1d1206bfb6e6c41f7d91b20b77e20f929fa6a7