pyrtlsdr/pyrtlsdr #107: error from the

Grekery, 2019-10-24:
when i import RtlSdr，i just get this

    from rtlsdr import RtlSdr
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "/usr/local/lib/python3.6/dist-packages/pyrtlsdr-0.2.91-py3.6.egg/rtlsdr/__init__.py", line 56, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "/usr/local/lib/python3.6/dist-packages/pyrtlsdr-0.2.91-py3.6.egg/rtlsdr/librtlsdr.py", line 61, in <module>
  File "/usr/lib/python3.6/ctypes/__init__.py", line 361, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 366, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: python3: undefined symbol: rtlsdr_get_device_count

how can i solve this problem？thanks！

closed 2019-10-24 by Grekery

nocarryr, 2019-10-24:
@Grekery were you able to find and resolve the issue?

Grekery, 2019-10-24:
yes I have solved it------------------ 原始邮件 ------------------

 
&mdash;