pyrtlsdr/pyrtlsdr #108: Using multiple SDR and threading

snowsquizy, 2019-12-13:
I have combined the PSD example with threading using kerberos SDR (4 SDR in a single package)
with the code below
********************************************
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
#
#  sdr_threading.py

import threading
import time
import pylab as plt
from rtlsdr import *

s_num = RtlSdr.get_device_serial_addresses()

class myThread (threading.Thread):
   def __init__(self, s_num, name):
      threading.Thread.__init__(self)
      self.s_num = s_num
      self.name = name
   def run(self):
      print ("Starting " + self.name)
      run_psd(s_num)
      print ("Exiting " + self.name)

def run_psd(serial_num):
    sdr = RtlSdr(serial_number=serial_num)
    sdr.sample_rate = 2.4e6
    sdr.center_freq = 95e6
    sdr.gain = 49
    samples = sdr.read_samples(2560*1024)
    sdr.close()

    # use matplotlib to estimate and plot the PSD
    plt.psd(samples, NFFT=2048, Fs=sdr.sample_rate/1e6, Fc=sdr.center_freq/1e6)
    plt.xlabel('Frequency (MHz)')
    plt.ylabel('Relative power (dB)')
    plt.title('SDR #{0}'.format(serial_num))

    plt.savefig('PSD_concurrent {0}.svg'.format(serial_num))
    plt.close()

thread1 = myThread(s_num[0], 'SDR 1')
thread2 = myThread(s_num[1], 'SDR 2')
thread3 = myThread(s_num[2], 'SDR 3')
thread4 = myThread(s_num[3], 'SDR 4')

tic = time.time()

thread1.start()
thread2.start()
thread3.start()
thread4.start()
thread1.join()
thread2.join()
thread3.join()
thread4.join()

toc = time.time()

print('Total Run Time : {0}'.format(toc-tic))

**************************************
However when run it throws the following error for all 4 SDR

-------------------------------------------------------
Exception in thread SDR 3:
Traceback (most recent call last):
  File "/home/andrew/anaconda3/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "sdr_threading.py", line 39, in run
    run_psd(s_num)
  File "sdr_threading.py", line 44, in run_psd
    sdr = RtlSdr(serial_number=serial_num)
  File "/home/andrew/anaconda3/lib/python3.5/site-packages/rtlsdr/rtlsdr.py", line 133, in __init__
    self.open(device_index, test_mode_enabled, serial_number)
  File "/home/andrew/anaconda3/lib/python3.5/site-packages/rtlsdr/rtlsdr.py", line 162, in open
    device_index = self.get_device_index_by_serial(serial_number)
  File "/home/andrew/anaconda3/lib/python3.5/site-packages/rtlsdr/rtlsdr.py", line 106, in get_device_index_by_serial
    result = librtlsdr.rtlsdr_get_index_by_serial(serial)
ctypes.ArgumentError: argument 1: <class 'TypeError'>: wrong type
------------------------------------------------------

I have tried it individually the PSD code example and they do work

nocarryr, 2019-12-13:
Can you check the return value of the serial addresses?
```python
s_num = RtlSdr.get_device_serial_addresses()
print(s_num)
```

Also, when you tried individually, were you using the same values for the `serial_number` argument?

snowsquizy, 2020-01-10:
The reason for the error is I was passing the serial number and not the device index for the serial number.  Hence the wrong type error.

closed 2020-01-10 by snowsquizy

gijsneerhof, 2020-11-12:
When using this, I get the error (intermittently) LIBUSB_ERROR_ACCESS (-3 ) insufficient permissions. 
This is because multiple threads try to access the USB at the same time, causing them to not have access. What I did to solve it, was to take out the opening of the SDR from `run_psd()` and instantiated the sdr objects before multithreading.