pyrtlsdr/pyrtlsdr #120: ImportError: Error loading librtlsdr. Make sure librtlsdr (and all of its dependencies) are in your path 64 bits

Bakira-cyber, 2020-11-24:
I added the rtlsdr.dll file in the rtl-sdr-release\x64.
[image]
But when I run the first example, it seems that the program doesn't find dll file.
I'm using a Python 64-bits. So I used a rtlsdr.dll 64 bits.

AYUSHBlaze, 2021-03-27:
Hey Bakira!, did you find a solution for this yet?

nocarryr, 2021-03-27:
The last time I tried working with librtlsdr in Windows, I was unable to get the x64 driver to work. It most likely has to do with the compilation method of the library (Python is pretty picky with Windows compilers).

I was able to get the x86 (32-bit) driver to work just fine though (using an x86 Python installation)

Martin-P, 2021-07-07:
Hi all,

I just installed this lib and got the 64 bit version working.

Visit https://osmocom.org/projects/rtl-sdr/wiki/Rtl-sdr#Windows and check the [Windows binaries](https://ftp.osmocom.org/binaries/windows/rtl-sdr/). These are more up-to-date than the GitHub mirrors (https://github.com/librtlsdr/librtlsdr and https://github.com/steve-m/librtlsdr). I downloaded [rtl-sdr-64bit-20210704.zip](https://ftp.osmocom.org/binaries/windows/rtl-sdr/rtl-sdr-64bit-20210704.zip) and unzipped the contents. I placed all its contents directly at my location `C:\Software\rtl-sdr-64bit-20210704` (so e.g. `librtlsdr.dll` is located at `C:\Software\rtl-sdr-64bit-20210704\librtlsdr.dll`) and added `C:\Software\rtl-sdr-64bit-20210704` to the environment variables.

The examples in the README.md work fine for me using the above method. Only found one issue in the example (#122), but that is not related to the ImportError.

I hope this helps anyone facing the same issue.

nocarryr, 2021-07-08:
@Martin-P great find!  I didn't realize osmocom was publishing up-to-date builds for windows.

These are based off of the osmo branch, aka the [steve-m fork](https://github.com/steve-m/librtlsdr) which isn't fully up-to-date with [librtlsdr](https://github.com/librtlsdr/librtlsdr) but still active.

I'll try and put this info into the project docs for others.  Thanks for finding this