pyrtlsdr/pyrtlsdr #127: invalid sample rate sent from client to server hangs server

Nostraquedeo, 2022-03-26:
After adding print statements all over the place I found that sending an invalid setting will cause the server to hang. The Linux port is still controlled by the process and shows the client connected. The client gets a 20-second timeout. I can not reconnect until I Control-C twice to kill the script and re-execute.

Besides the obvious "Well don't do that" how can this fail gracefully with a refusal or warning?

Nostraquedeo, 2022-03-27:
After more searching, the problem appears to be twofold in the rtlsdr.py file.
The LibUSBError dict did not have a listing for -22.
And the self.close() appears to freeze the server.

By adding an entry for error code -22 and commenting out the self.close() the server continues to operate.
I have not figured out how to send a message back to the client yet informing of the failure to set the setting.
But I can always run a get_setting call to confirm the setting was taken.

nocarryr, 2022-03-27:
Tbh, the tcp module needs a serious overhaul, but the immediate issue is there are no exception handlers in place for the parts that actually control the device:

https://github.com/roger-/pyrtlsdr/blob/64835e778a59d6a887cbe1a7f53e33f473614adc/rtlsdr/rtlsdrtcp/server.py#L191-L194

and

https://github.com/roger-/pyrtlsdr/blob/64835e778a59d6a887cbe1a7f53e33f473614adc/rtlsdr/rtlsdrtcp/server.py#L204

should be wrapped in a `try: ... except:` block and send a `CommunicationError` back to the client.

I honestly didn't think many people were using that code, so I haven't done much with it.  I'll try and work on that when time allows

Nostraquedeo, 2022-03-27:
Thanks for the quick review.  And thanks  for all your hard work on this.

nocarryr, 2026-11-13:
Addressed in #272.  Feel free to re-open if the issue persists once it's released in the next release (>0.5.0)

closed 2026-11-13 by nocarryr