pyrtlsdr/pyrtlsdr #129: Add test for bias t function

bradylangdale, 2022-07-27:
- Missing test for the bias t control due to merge of #123

nocarryr, 2022-07-27:
@bradylangdale in order to test in a containerized environment, I had to create a mock of the librtlsdr wrapper.

It may be a bit confusing to understand what to put where, but if you feel confident, be my guest.
 
You'll have to add the new library call as a method of [this class](https://github.com/roger-/pyrtlsdr/blob/1a857a45d63dcb9ae32f9883d867c963e82fb046/tests/testlibrtlsdr.py#L22) as well an internal state attribute.

Then the actual test code could probably go in the `generic_test` [here](https://github.com/roger-/pyrtlsdr/blob/1a857a45d63dcb9ae32f9883d867c963e82fb046/tests/utils.py#L88)

nocarryr, 2022-07-27:
After looking at the librtlsdr header, I see there is no "getter" for bias-t, so ignore what I said about internal state above