pyrtlsdr/pyrtlsdr #131: AttributeError: function 'rtlsdr_set_bias_tee' not found

ImNiceButActuallyNot, 2022-09-27:
I've tried to install pyrtlsdr according to this [tutorial](https://www.youtube.com/watch?v=XT9pybErVW8).
When I import the library there is an error "function 'rtlsdr_set_bias_tee' not found" (the error is the same when the rtl is connected or disconnected).

system information:
- I use windows 10
- my rtl is RTL2832

I will be glad if you could help me.

nocarryr, 2022-10-02:
I haven't yet watched the video you linked, but it would seem like you're using an older version of librtlsdr.  Can you try using one of the librtlsdr releases found here:
https://github.com/librtlsdr/librtlsdr/releases

ImNiceButActuallyNot, 2022-10-04:
Thanks for the response.
The video is irrelevant, I've found that the reason the library didn't work was that the pip installer didn't work.
When using the setup script in the master branch everything worked fine 😊

Happy to be able to use this tool now🙃

closed 2022-10-04 by ImNiceButActuallyNot

nocarryr, 2022-10-05:
@ImNiceButActuallyNot glad you were able to get things working.

I'm concerned though about possible packaging issues so I'd like to investigate a little further.  Can you tell me anything about your python installation?  Do you have more than one version installed?  And are you using virtualenv or did you do a user install of pyrtlsdr?

Was the pip install something like this?

```bash
pip install --user pyrtlsdr
```

Or like this:
```bash
python -m pip install --user pyrtlsdr
```

Before you installed from `setup.py`, did you uninstall? 
```bash
pip uninstall pyrtlsdr
```

If you didn't do that first, did you get errors when you ran `python setup.py install` ?

ImNiceButActuallyNot, 2022-10-07:
I have only python3.7 on my computer.

The pip install was like the first option `pip install pyrtlsdr`.

Before running `setup.py` I've uninstalled the library (`pip uninstall pyrtlsdr`).

nocarryr, 2022-10-07:
@ImNiceButActuallyNot thanks!  I'll investigate the windows packaging

gmrwvu, 2023-07-19:
I also ran into this error. What solved it for me was deleting an rtlsdr.dll that was in the directory.