pyrtlsdr/pyrtlsdr #132: Trying to run Test Code - Getting Error

loganfarrow, 2022-10-01:
M1 Mac Pro 
Python 3.9
Ran pip install pyrtlsdr 

Then tried the test code to print examples and returned 
"ImportError: cannot import name 'RtlSdr' from partially initialized module 'rtlsdr' (most likely due to a circular import)"
Any help?

nocarryr, 2022-10-02:
@loganfarrow which example code are you using?  Is it [demo_waterfally.py](https://github.com/pyrtlsdr/pyrtlsdr/blob/7a854dcf5a1c30d0e29014ae9f24e3b35a7af859/demo_waterfall.py)?

If so, did you download only that file or the entire repository?

If you downloaded the repository AND installed pyrtlsdr with pip, there would be two modules named "rtlsdr" at import time.
Simply deleting the "rtlsdr" folder should solve that issue.

loganfarrow, 2022-10-02:
I ran this after only doing “pip install pyrtlsdr” and got the circular import error. 
I had no issues getting it imported on Linux so not sure what went wrong back on my mac.  

from rtlsdr import RtlSdr

sdr = RtlSdr()

# configure device
sdr.sample_rate = 2.048e6  # Hz
sdr.center_freq = 70e6     # Hz
sdr.freq_correction = 60   # PPM
sdr.gain = 'auto'

print(sdr.read_samples(512))

nocarryr, 2022-10-02:
Ok, since I'm in front of a mac at the moment, I installed librtlsdr using [homebrew](https://formulae.brew.sh/formula/librtlsdr#default) and pyrtlsdr via pip.

Since I don't have a dongle with me, I'm obviously not able to fully test, but I get past the initial import:

```python
Python 3.9.13 (main, Sep 11 2022, 11:45:59) 
[Clang 10.0.0 (clang-1000.10.44.4)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from rtlsdr import RtlSdr
>>> sdr = RtlSdr()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/production/.local/share/virtualenvs/production-LedfrSRR/lib/python3.9/site-packages/rtlsdr/rtlsdr.py", line 133, in __init__
    self.open(device_index, test_mode_enabled, serial_number)
  File "/Users/production/.local/share/virtualenvs/production-LedfrSRR/lib/python3.9/site-packages/rtlsdr/rtlsdr.py", line 171, in open
    raise LibUSBError(result, 'Could not open SDR (device index = %d)' % (device_index))
rtlsdr.rtlsdr.LibUSBError: <LIBUSB_ERROR_IO (-1): Input/output error> "Could not open SDR (device index = 0)"
```

That's the expected result.  In your case, you can't get past the initial import.  I'm wondering whether or not your current directory is clean.  What happens if you change directories (make an empty folder?)  and run the same code?

- Mac Pro (Late 2013)
- v10.13.5 (High Sierra)
- Python 3.9.13

loganfarrow, 2022-10-02:
@nocarryr Okay that helped. Made a new directory and did a test run with the same code and now I'm getting that rtlsdr does not exist. 

Going back to terminal "pip3 install pyrtlsdr" tells me that it is already installed. Is this some sort of path error you can point me to a way to fix?

Thank you for all your help.

nocarryr, 2022-10-02:
Does the error say that `rtlsdr` doesn't exist or is it `librtlsdr`?

There's a pretty big difference there.

If you see:
`Error loading librtlsdr. Make sure librtlsdr (and all of its dependencies) are in your path`
then there's an issue with the installation of [librtlsdr](https://github.com/librtlsdr/librtlsdr).

But if you get something like:
`ModuleNotFoundError: No module named 'rtlsdr'`
it's got something to do with your pip installation.

You may be calling a different `python` executable (and version) than what you used for the pip install.  Try to launch it specifically with `python3.9` instead of `python`.

If it still throws `ImportError` or `ModuleNotFoundError`, try installing it this way:

```bash
python3.9 -m pip install pyrtlsdr
```

This forces the install location for that specific py version

closed 2023-07-22 by loganfarrow