pyrtlsdr/pyrtlsdr #135: import pyrtlsdr No Module named 'pyrtlsdr'

JAbelP, 2023-01-06:
`import subprocess
import sys

def install(package):
    subprocess.check_call([sys.executable, "-m", "pip", "install", package])

try:
    import pyrtlsdr
except:
    install("pyrtlsdr")
    import pyrtlsdr
`

(Radio) D:\Code\Python\radio>python listen_to_fm.py
Requirement already satisfied: pyrtlsdr in d:\code\python\radio\lib\site-packages (0.2.93)
WARNING: You are using pip version 21.2.4; however, version 22.3.1 is available.
You should consider upgrading via the 'D:\Code\Python\radio\Scripts\python.exe -m pip install --upgrade pip' command.
Traceback (most recent call last):
  File "D:\Code\Python\radio\listen_to_fm.py", line 8, in <module>
    import pyrtlsdr
ModuleNotFoundError: No module named 'pyrtlsdr'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\Code\Python\radio\listen_to_fm.py", line 11, in <module>
    import pyrtlsdr
ModuleNotFoundError: No module named 'pyrtlsdr

I am on a windows ten machine running python 3.10.2, I can install and use other libraries without any issue I saw I may be missing decencies, but because the error looks different from those that are missing the decencies I don't know if something else is wrong.

(I am also not sure where I would put them in this file, the documentation mentions putting it next to a README but I can't find one in the file tree) 
[image]

nocarryr, 2023-01-09:
The installation seems to have worked properly, but the import should be `import rtlsdr` instead of `import pyrtlsdr`

nocarryr, 2023-08-02:
@JAbelP  Check out the latest release (0.3.0).  Installation of `librtlsdr` can now be done via pip (instructions in the README)

Please drop a comment if it works or not.  I'd appreciate the feedback