pyrtlsdr/pyrtlsdr #139: Changing `sdr.center_freq` while streaming.

hfwang132, 2023-05-31:
Hello, I am working on an example for streaming FM audio. So far, I have had the FM demodulation part working, and I can hear the audio with a center frequency pre-adjusted to that of a radio station. Here is the implementation:

```python
import ipywidgets as widgets

import nest_asyncio
nest_asyncio.apply()
import asyncio

sdr = RtlSdr()

sdr.sample_rate = 2_400_000 
sdr.center_freq = 92_700_000 # an FM radio station
sdr.gain = 4

it_num = 10

def fm_demodulation(data):
    # some algorithms
    # ...
    return audio

async def streaming():
    it = 0
    async for data in sdr.stream(num_samples_or_bytes=1024*3000, format='bytes', loop=None):        
        # perform FM demodulation
        audio = fm_demodulation(data)
        # play the audio
        display(Audio(audio, autoplay=True, rate=48000, normalize=False))
        
        it = it + 1
        if it == it_num:
            break
        
async def main():
    print("task started")
    task = asyncio.create_task(streaming())
    await task
    print("task finished")

asyncio.run(main())
```

Now, I want to adjust the frequency in real time via the `ipywidgets` in a Jupyter Notebook:

```python
import ipywidgets as widgets

import nest_asyncio
nest_asyncio.apply()
import asyncio

sdr.sample_rate = 2_400_000 
sdr.center_freq = 92_700_000 # an FM radio station
sdr.gain = 4

it_num = 10

async def streaming():
    it = 0
    async for data in sdr.stream(num_samples_or_bytes=1024*3000, format='bytes', loop=None):        
        # perform FM demodulation
        audio_samples = fm_demodulation(data)
        # play the audio
        display(Audio(audio, autoplay=True, rate=48000, normalize=False))
        
        it = it + 1
        if it == it_num:
            break
            
def on_slider_change(change):
    new_center_freq = change['new']
    # Update the sdr.center_freq parameter with the new value
    sdr.center_freq = new_center_freq
    with out:
        print("sdr.center_freq is changed to %d" % sdr.center_freq)
        
async def main():
    print("task started")
    task = asyncio.create_task(streaming())
    await task
    print("task finished")

slider = widgets.IntSlider(
    value=92_700_000,  # Initial value
    min=80_000_000, 
    max=110_000_000, 
    step=100_000, 
    description='Center Freq:',
    orientation='horizontal',
)

out = widgets.Output()

slider.observe(on_slider_change, names='value')

display(slider, out)
    
asyncio.run(main())
```

However, the `streaming()` coroutine, although performed asynchronously, seems to block the operation of changing `sdr.center_freq` via the slider:

[image]

As seen above, I am able to change `sdr.center_freq` via the slider. But I am only able to change it after all the audios are streamed, and those audios are streamed with `sdr.center_freq` being at its initial state.

Therefore, I am writing to ask if there is a proper way to change `center_freq` in real-time, so that a simple FM radio receiver application can be achieved?

I know that this question is not quite related to `pyrtlsdr` itself, but I thought it would be a great example of using `pyrtlsdr`. I would be grateful if anybody could give some light on this. Thanks!

hfwang132, 2023-05-31:
Thank you very much @nocarryr ! I really appreciate your reply.

When I execute your code, it gets stuck at `center_freq = await retune_queue.get()`. I will look into it further tomorrow. 

---

By the way, I found that maybe the problem is not related to `pyrtlsdr` at all, but rather lies in the nature of `ipywidgets`. It seems that `ipywidgets` will only update the widget values after the execution of all cells is finished. Consider the following code:

```python
import ipywidgets as widgets

slider = widgets.IntSlider(
    value=92_700_000,  # Initial value
    min=80_000_000, 
    max=110_000_000, 
    step=100_000, 
    description='Center Freq:',
    orientation='horizontal',
)

display(slider)

for i in range(10):
    print(f"second {i}: slider.value is {slider.value}")
    time.sleep(1)
```

Even though I slide the slider while the loop is running, the printed values won't change:

```
second 0: slider.value is 92700000
second 1: slider.value is 92700000
second 2: slider.value is 92700000
second 3: slider.value is 92700000
second 4: slider.value is 92700000
second 5: slider.value is 92700000
second 6: slider.value is 92700000
second 7: slider.value is 92700000
second 8: slider.value is 92700000
second 9: slider.value is 92700000
```

The same is true if `asyncio` is used:

```python
import nest_asyncio
nest_asyncio.apply()
import asyncio

async def main():
    for i in range(10):
        print(f"second {i}: slider.value is {slider.value}")
        await asyncio.sleep(1)

asyncio.run(main())
```

In other words, any running cells would totally block the widgets from getting updated.

Therefore, by far, maybe the problem isn't really related to `pyrtlsdr`, but rather `ipywidgets`. I will consider developing the web app from a lower level.

nocarryr, 2023-05-31:
Having asyncio behave with any UI framework is rarely fun or easy since they all depend on the main thread along with an internal event loop.

Looks like they have an async mode though:
https://ipywidgets.readthedocs.io/en/stable/examples/Widget%20Asynchronous.html

hfwang132, 2023-06-01:
You are right, and I have tried the solutions mentioned in https://ipywidgets.readthedocs.io/en/stable/examples/Widget%20Asynchronous.html before, but unfortunately the whole loop would be blocked by `wait_for_change()`.

However, today I found this amazing library: https://pypi.org/project/jupyter-ui-poll/, which allows me to poll UI events. Finally, I get my app working properly, and I can adjust the `sdr.center_freq` in real time and hear the difference. The final code is as follows:

```python
from jupyter_ui_poll import ui_events

sdr.sample_rate = 2_400_000 
sdr.center_freq = 92_700_000 # an FM radio station
sdr.gain = 4
it_num = 10

async def streaming():
    it = 0
    async for data in sdr.stream(num_samples_or_bytes=1024*3000, format='bytes', loop=None):        
        # perform FM demodulation
        audio = fm_demodulation(data)
        # poll UI events
        with ui_events() as poll:
            poll(10)
            sdr.center_freq = slider.value
        # play the audio
        display(Audio(audio/2, autoplay=True, rate=48000, normalize=False))
        
        it = it + 1
        if it == it_num:
            break

slider = widgets.IntSlider(
    value=92_700_000,  # Initial value
    min=80_000_000, 
    max=110_000_000, 
    step=100_000, 
    description='Center Freq:',
    orientation='horizontal',
)

display(slider)
    
asyncio.run(streaming())
```

I am surprised that it works so well. I guess in `librtlsdr`, you are actually allowed to change the center frequency during receiving samples.

Anyway, thank you for your attention in this matter @nocarryr . I guess that will put an end to my problem. It would be a cute example of using `pyrtlsdr` to make a simple FM receiver in Jupyter Notebook.

hfwang132, 2023-06-01:
By the way, every time an audio np.array is played, there will be a playback progress bar showing up, which is kind of messy. One can hide them using the following code:

```python
from IPython.display import HTML, display
# hide the progress bar of IPython.display.Audio
hide_audio_css = """
<style>
audio { display: none }
</style>
"""
display(HTML(hide_audio_css))
```

nocarryr, 2023-06-01:
That's interesting honestly, I've never thought to try changing center frequencies while reading samples/bytes, so I never would've guessed.

I'm assuming the `ui_events` context manager and its `poll` method are non-blocking?  Interesting that it's not an async context manager (`async with`).  I'll have to look into that library for my own purposes, thanks!

Are we good to close this issue at present?

hfwang132, 2023-06-01:
Yes, it is okay to close the issue now, since the problem is resolved. I guess there is a lot for me to learn about concurrence of Python/IPython. Again, many thanks!

closed 2023-06-01 by hfwang132