pyrtlsdr/pyrtlsdr #140: "No module named 'pyrtlsdr'

cjcfojc, 2023-06-13:
some info off the bat...

Ubuntu Mate 22.04
5.19.0-43-generic
(currently) Python 3.10

This is on a pretty fresh install...nothing but SDR related things and mapping tools, but a few days ago on my pre-wiped machine, i ran into this same problem and elected to do a fresh install to try and mitigate the issue, but alas the issue persists.  im not ultra savvy with python, or even python related CLI, so i'll admit to enlisting the help of chatgpt at least this time around (it was much quicker to exasperate my options the second time).  this is the entire terminal output from the process + various troubleshooting steps.  One other thing i tried that isn't shown here, is i tried with Python 3.9.  same result.  any help would be appreciated

[terminal_output.txt](https://github.com/pyrtlsdr/pyrtlsdr/files/11729179/terminal_output.txt)

cjcfojc, 2023-06-24:
no reply to this issue after two weeks?

nocarryr, 2023-06-24:
For future reference, it's much more helpful to paste your terminal output into the comment using a [markdown codeblock](https://www.markdownguide.org/extended-syntax/#fenced-code-blocks) like this:

```bash
carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ python3 -m venv env
carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ source env/bin/activate
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ python SpecSDR.py
['/home/carmen/Documents/Misc Scripts/SpecSDR', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/home/carmen/Documents/Misc Scripts/SpecSDR/env/lib/python3.10/site-packages']
Traceback (most recent call last):
  File "/home/carmen/Documents/Misc Scripts/SpecSDR/SpecSDR.py", line 8, in <module>
    from PIL import Image
ModuleNotFoundError: No module named 'PIL'
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ pip install pillow
Collecting pillow
  Downloading Pillow-9.5.0-cp310-cp310-manylinux_2_28_x86_64.whl (3.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.4/3.4 MB 15.3 MB/s eta 0:00:00
Installing collected packages: pillow
Successfully installed pillow-9.5.0
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ python SpecSDR.py
['/home/carmen/Documents/Misc Scripts/SpecSDR', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/home/carmen/Documents/Misc Scripts/SpecSDR/env/lib/python3.10/site-packages']
Traceback (most recent call last):
  File "/home/carmen/Documents/Misc Scripts/SpecSDR/SpecSDR.py", line 9, in <module>
    from pyrtlsdr import RtlSdr
ModuleNotFoundError: No module named 'pyrtlsdr'
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ pip install pyrtlsdr
Collecting pyrtlsdr
  Using cached pyrtlsdr-0.2.93-py2.py3-none-any.whl (25 kB)
Installing collected packages: pyrtlsdr
Successfully installed pyrtlsdr-0.2.93
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ python SpecSDR.py
['/home/carmen/Documents/Misc Scripts/SpecSDR', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/home/carmen/Documents/Misc Scripts/SpecSDR/env/lib/python3.10/site-packages']
Traceback (most recent call last):
  File "/home/carmen/Documents/Misc Scripts/SpecSDR/SpecSDR.py", line 9, in <module>
    from pyrtlsdr import RtlSdr
ModuleNotFoundError: No module named 'pyrtlsdr'
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ python3.10 SpecSDR.py
['/home/carmen/Documents/Misc Scripts/SpecSDR', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/home/carmen/Documents/Misc Scripts/SpecSDR/env/lib/python3.10/site-packages']
Traceback (most recent call last):
  File "/home/carmen/Documents/Misc Scripts/SpecSDR/SpecSDR.py", line 9, in <module>
    from pyrtlsdr import RtlSdr
ModuleNotFoundError: No module named 'pyrtlsdr'
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ pip install pyrtlsdr
Requirement already satisfied: pyrtlsdr in ./env/lib/python3.10/site-packages (0.2.93)
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ python -c "import pyrtlsdr"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'pyrtlsdr'
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ pip install --upgrade --force-reinstall pyrtlsdr
Collecting pyrtlsdr
  Using cached pyrtlsdr-0.2.93-py2.py3-none-any.whl (25 kB)
Installing collected packages: pyrtlsdr
  Attempting uninstall: pyrtlsdr
    Found existing installation: pyrtlsdr 0.2.93
    Uninstalling pyrtlsdr-0.2.93:
      Successfully uninstalled pyrtlsdr-0.2.93
Successfully installed pyrtlsdr-0.2.93
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ python3.10 SpecSDR.py
['/home/carmen/Documents/Misc Scripts/SpecSDR', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/home/carmen/Documents/Misc Scripts/SpecSDR/env/lib/python3.10/site-packages']
Traceback (most recent call last):
  File "/home/carmen/Documents/Misc Scripts/SpecSDR/SpecSDR.py", line 9, in <module>
    from pyrtlsdr import RtlSdr
ModuleNotFoundError: No module named 'pyrtlsdr'
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ pip install pyrtlsdr --user
ERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ sudo apt-get install librtlsdr-dev
[sudo] password for carmen: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
librtlsdr-dev is already the newest version (0.6.0-4).
The following package was automatically installed and is no longer required:
[... 173 lines cut ...]
['/home/carmen/Documents/Misc Scripts/SpecSDR', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/home/carmen/Documents/Misc Scripts/SpecSDR/env/lib/python3.10/site-packages', '/home/carmen/Documents/Misc Scripts/SpecSDR/env/lib/python3.10/site-packages/pyrtlsdr-0.2.93-py3.10.egg']
Traceback (most recent call last):
  File "/home/carmen/Documents/Misc Scripts/SpecSDR/SpecSDR.py", line 9, in <module>
    from pyrtlsdr import RtlSdr
ImportError: cannot import name 'RtlSdr' from 'pyrtlsdr' (unknown location)
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ pip install --force-reinstall pyrtlsdr==0.2.0
Collecting pyrtlsdr==0.2.0
  Using cached pyrtlsdr-0.2.0.zip (12 kB)
  Preparing metadata (setup.py) ... done
[... cut ...]
['/home/carmen/Documents/Misc Scripts/SpecSDR', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/home/carmen/Documents/Misc Scripts/SpecSDR/env/lib/python3.10/site-packages', '/home/carmen/Documents/Misc Scripts/SpecSDR/env/lib/python3.10/site-packages/pyrtlsdr-0.2.93-py3.10.egg']
Traceback (most recent call last):
  File "/home/carmen/Documents/Misc Scripts/SpecSDR/SpecSDR.py", line 9, in <module>
    from pyrtlsdr import RtlSdr
ImportError: cannot import name 'RtlSdr' from 'pyrtlsdr' (unknown location)
(env) carmen@thevault:~/Documents/Misc Scripts/SpecSDR$ 
```

nocarryr, 2023-06-24:
From looking at a few of your installation attempts, it's possible it could be an issue with the `pip` command not being mapped correctly (even after creating the virtual environment).

You might want to try this:
```bash
python -m pip install pyrtlsdr
```

If that's the case, you can fix the `pip` execution like this:

```bash
python -m pip install -U pip
```

cjcfojc, 2023-06-25:
> From looking at a few of your installation attempts, it's possible it could be an issue with the `pip` command not being mapped correctly (even after creating the virtual environment).
> 
> You might want to try this:
> 
> ```shell
> python -m pip install pyrtlsdr
> ```
> 
> If that's the case, you can fix the `pip` execution like this:
> 
> ```shell
> python -m pip install -U pip
> ```

if you look through the terminal output, you'll see that both of those solutions have already been attempted.  All of this has been done on a fresh install of Mate 22.04, as well as Mate Pi 22.04 ARM64 and the result is the same.  all other python modules that i use and require install correctly and function as intended.  pyrtlsdr is the only one that does this.

cjcfojc, 2023-06-26:
Confirmed same situation this morning on a brand new Raspberry Pi 4 running a fresh Ubuntu Mate Pi 22.04 arm64.  All manners of installing pyrtlsdr, including from source, yield the same result.  So the following installs yield the same result...Ubuntu Mate 22.04, Ubuntu Mate Pi 22.04 arm64, "regular" Ubuntu 22.04.

nocarryr, 2023-08-02:
@cjcfojc I can't believe I missed this, but I just noticed when going through issues (for other purposes)....

The import is your issue.  The package name should be `rtlsdr`, not `pyrtlsdr`

```python
import rtlsdr
```
or 
```python
from rtlsdr import RtlSdr
```

The project name (as far as pip, etc is concerned) is different than the package name that's used for import

closed 2023-08-07 by cjcfojc