pyrtlsdr/pyrtlsdr #144: Problem with windows 11?  Error loading librtlsdr

witc, 2023-07-20:
Hi, I used pyrtlsdr on windows 10 and now I want to use  it on another machine with windows 11, but error appears when I try to import _`from rtlsdr import *`_
_**Error loading librtlsdr. Make sure librtlsdr (and all of its dependencies) are in your path**_
rtl_sdr.exe seems to work, Is there any idea how to solve it?
I installed it like _pip install pyrtlsdr_  with succes.

witc, 2023-07-23:
Can it be in conflict with installed GNU radio?

witc, 2023-07-23:
I have read in trubleshootin in this PDF: https://pyrtlsdr.readthedocs.io/_/downloads/en/latest/pdf/ : I should check if **rtl_sdr.exe** works, but I can even find the **rtl_sdr.exe**. 
I did:
pip show _pyrtlsdr_ - it showed me the path  C:\Python_3_11\Lib\site-packages
There are 2 folders rtlsdr and pyrtlsdr-0.2....... only in rtlsdr are some python files, but no any **rtl_sdr.exe** and even can not see the README
_**_Windows: Make sure all the librtlsdr DLL files (librtlsdr.dll, libusb-1.0.dll) are in your system path, or the
same folder as this README file_**_

nocarryr, 2023-07-23:
dll loading in Windows is a bit finicky tbh.  Check to see if [ctypes](https://docs.python.org/3/library/ctypes.html#ctypes-reference) is able to find the library:

```python
from ctypes.util import find_library

print(find_library('rtlsdr')
print(find_library('librtlsdr')
```
If no path is returned from either of those, the dll's aren't in any of the `%PATH%` directories (or they're built for the wrong architecture I think).

If you know the exact filename of the dll, you can try loading it directly:
```python
from ctypes import CDLL

dll = CDLL('<full-path-to-dll-file>')
```
If it fails, you may get more helpful information from the exception raised.

I started working on a "sister project" called [pyrtlsdrlib](https://github.com/pyrtlsdr/pyrtlsdrlib) to reduce the number of issues users have with this, but it's nowhere near completion.  I think it's about time I finish it

witc, 2023-07-23:
Thank you. No path was returned. I just wonder how this all works. Is that OK?
1) **pyrtlsdr** does nothing but wraps the **librtlsdr.dll**
2) librtlsdr should be downloaded from this website https://github.com/librtlsdr/librtlsdr/releases I used **_rtlsdr-bin-w64_dlldep_**
3) from this I took _librtlsdr.dll_ and put it into _**C:\Windows\SysWOW64**_ also I added there downloaded  **_libusb-1.0.dll_**

But probably it is not right - because I have still the same error

nocarryr, 2023-07-23:
If you place the dll files in the current working directory of your application (or where you're running commands from), it *should* work.
This still isn't an ideal solution, but library linking in windows is far from ideal to begin with.

And for `#1`, yes, you're correct.  This is just a python wrapper for the actual library, so it does nothing without the library (dll) itself.

nocarryr, 2023-08-02:
@witc Check out the latest release (0.3.0).  Installation of `librtlsdr` can now be done via pip (instructions in the README)

Please drop a comment if it works or not.  I'd appreciate the feedback

witc, 2024-04-04:
> @witc Check out the latest release (0.3.0). Installation of `librtlsdr` can now be done via pip (instructions in the README)
> Please drop a comment if it works or not. I'd appreciate the feedback
Hi,
I attempted to understand how you suggested I could resolve this. I did it as shown below, but it's not functional, I probably made a mistake?
[image]
[image]

Niyusha, 2024-07-12:
I'm having a similar issue on Windows 10. 

The IDE I'm using is PyChrm and even thought I import both the pyrtlsdr library and librtlsdr the import does not get resolved correctly.

georgeh1ll, 2024-12-17:
Hi, I managed to fix it on windows using
`pip install setuptools`
(possibly with sudo in linux case).