pyrtlsdr/pyrtlsdr #147: Read samples with timeout

witc, 2023-09-14:
Hi,

Is it possible to read samples with timeout? I have problem the when I am periodically reading samples from SDR and someone disconnect the RTLSDR USB device from PC, then my program gets stuck in function sdr.read_samples().

I just want something like sdr.read_samples(num_samples=1024, timeut = 0.5 sec)

I tried:
```
 p = multiprocessing.Process(target=self.sdr.read_samples())
        p.start()
        p.join(2)
        if p.is_alive():
            p.terminate()
            # OR Kill - will work for sure, no chance for process to finish nicely however
            # p.kill()
            p.join()
```

Also tried something with subprocess with timeout, but nothing worked reliably

Any Idea please? Thank you

nocarryr, 2023-09-16:
TBH, I'm not sure that it's possible to rescue a read operation that way.  If the device is unplugged in the middle of a read, I can't imagine how to determine which of the samples in the buffer were actually read.  The samples that were missed would likely be garbage since the entire buffer (`sizeof(uint8) * 1024 * 2`) is preallocated in the C library.

The only way to avoid the process crashing completely would be to catch the exception...

```python
import traceback

def read_samples(sdr: RtlSdr, num_samples: int = 1024) -> Sequence[int]:
    try:
        result = sdr.read_samples(num_samples)
    except OSError:
        # just return an empty ndarray or list if it fails.  could do None instead
        result = []
        # since we're ignoring the error, at least show it on stdout
        traceback.print_exc()
    return result

```

The result returned on error would have to be checked in your multiprocessing code, but you get the idea.  Be careful when throwing away exceptions though, you could wind up in a nasty loop if you don't handle the results properly

witc, 2023-09-18:
Thank you, but it seems like it's not working correctly. All I basically want is the function IsAnyRTLSDRConnectedToPC() - which returns True or False. But this method very often get stucked..