pyrtlsdr/pyrtlsdr #149: Processing the continuous stream

bigalnz, 2023-12-06:
I have a application where I need to process the stream continuously. I can not afford to miss any pieces of data as I am using the sample rate to do time calculations and calculate interval between pulses (even across processing samples)

Being new to this, I am unsure what I can do to optimise things from here, and wondered if the folks here could provide some code review/input. Apologies if not allowed here:

`
# Configure RTL-SDR parameters
sdr = RtlSdr()
sdr.sample_rate = 2.4e6  # 2.4 MHz
sdr.center_freq = 160270968 # MHz
sdr.gain = 'auto'

# Number of samples to read in each iteration
samples_to_process = 4.e6
threshold = 0.6
freq_offset = -43.8e4 #Hz
`
And then:
`
try:
    while True:
        samples = sdr.read_samples(samples_to_process)
        processed_data = process_samples(samples, sdr.sample_rate, freq_offset, threshold)
`        `

The actual processing block looks like:

`
def process_samples(samples, sample_rate, freq_offset, threshold):

    global index
    global lastrise
    global last
    global lastfalling
    global time_between_rising_edge
    global rssi
    global pulse_width

    t = np.arange(len(samples))/sample_rate
    samples = samples * np.exp(2j*np.pi*t*freq_offset)
    h = signal.firwin(501, 0.02, pass_zero=True)
    samples = np.convolve(samples, h, 'valid')
    samples = samples[::100]
    sample_rate = sample_rate/100
    samples = np.abs(samples)
    samples = np.convolve(samples, [1]*10, 'valid')/10
    max_samp = np.max(samples)
    samples /= np.max(samples)

    # plt.plot(samples)
    # plt.show()

    def islow(v):
        return v < threshold
    def ishigh(v):
        return v >= threshold

    # do nothing if no signals above threshold
    if np.max(samples) < threshold:
        return
    
    # samples array index(j), value(n) 
    for j,n in enumerate(samples):
        i = index + j
        # i = j        
        # Is this a rising edge?
        if ishigh(n) and islow(last):
            # then check is second rising edge? i.e. lastrise!=0
            if lastrise:
                time_between_rising_edge = round(sample_rate/(i-lastrise)*60,2)
                print(f"BPM: {time_between_rising_edge}")
            lastrise = i
        # Is this a falling edge?
        elif islow(n) and ishigh(last):
            pulse_width =  round((i-lastrise)/sample_rate,4)
            lastfalling = i      
            rssi = round(np.mean(samples[j-(i-lastrise):j]) * max_samp,2)
            print(f"rssi : {rssi}")
        last = n
    index = index + len(samples)
    `
    Any missing chunks will affect the rolling index.
    
Any feedback appreciated.

nocarryr, 2023-12-07:
As far as the sampling is concerned, using `async` mode would be ideal since it relies on the low-level C library for collection (then fires a callback with the data).  That way you're not relying on repeated calls from Python -> C to read and buffer samples.  That would of course complicate things in your overall code structure if you aren't already familiar with `asyncio`.

In your processing routine though, I can see a few things that could be easily optimized and some that I'd have to look at more to understand what you're doing...

You're recomputing things in every call to `process_samples()` that could be pre-computed at the module-level:

- `t = np.arange(len(samples))/sample_rate`
  - shouldn't change since it's based off of `samples_to_process`
- `h = signal.firwin(501, 0.02, pass_zero=True)`
  - Your filter coefficients don't need to be calculated every time, or do they?
- Is that a linear phase term I see (`np.exp(2j*np.pi*t*freq_offset`)?  Intriguing, I love those (unless I'm seeing it wrong)
- 'np.convolve' is slow compared to scipy.signal's version (unless they've changed it recently) and you're doing two convolutions (one on the full sample set and one on the decimated samples).  You might want to look at [scipy.signal.decimate](https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.decimate.html#scipy.signal.decimate) or some of the other goodies in scipy.signal

And yeah, I think you'll want to look at the `np.argmax` stuff and maybe some of the other indexing and searching routines to help vectorize your operations.  Always best to avoid for loops when you can, but I don't know of a great way to handle the pulse width calculations without iteration (at least not without lots of time, thought and experimentation lol)

Hope this helps

bigalnz, 2023-12-07:
In fact would you be interested in some paid tutoring on it?

nocarryr, 2023-12-07:
I'd start with the first two since they're the easiest. May not make a huge difference, but it'd be worth the minimal effort.

The `convolve -> decimate -> convolve` step would probably knock off a decent chunk if you can achieve the same thing with the scipy routines.

As far as sample count, it really just depends. The more stuff you read, the longer it'll take to process. And that time is spent while not reading more samples. You can always process on another thread and process in batches, but if your code can't keep up you'll start falling behind "wall time" anyways.  It's hard to say to be honest since it depends on what you're doing with the samples and takes a lot of experimentation and tweaking 

Oh... And most importantly, reading a lot of other people's code haha. Just not mine, it's not always the best

bigalnz, 2023-12-07:
Any idea what the valid values are for gain for a RTLSDR?

bigalnz, 2023-12-07:
Also something odd is going on - my signal should be low (approx < 0.1) then when it goes high about 0.9 - but when I get a chunk with no high states in it, its showing values about 0.8 - and this is triggering my threshold.