pyrtlsdr/pyrtlsdr #150: Sample size selection, np.complex128 vs np.complex64?

dkozel, 2023-12-24:
Hi, I noticed that the sample output format is using 64 bit floating point values for I and Q. Given the RTLSDR only outputs 8 bit integers why was 64 bit floats chosen rather than the more common 32 bit floats?
https://github.com/pyrtlsdr/pyrtlsdr/blob/b5069f9a9125169ada8b11189172070497b4f72f/rtlsdr/rtlsdr.py#L525

This is increasing both the memory requirements for buffering/storing samples as well as the computational complexity of operations on the samples.

roger-, 2023-12-24:
Good question, either we weren't unaware of numpy's csingle/complex64 or it didn't exist back then?

Would be a simple change to explore.

nocarryr, 2023-12-24:
I believe that was chosen as it's the default for complex arrays on most platforms.

I agree that it does seem wasteful, but it hasn't been brought up before now.  In order to avoid breaking existing user code though, I think this would need to be a configurable parameter (with the default being as-is).

I'm fairly certain numpy's type promotion would handle things in most cases, but there may be some edge cases where it wouldn't

dkozel, 2023-12-26:
Many other SDR libraries and applications are using 32 bit floats as their default data type, including Universal Radio Hacker, GNU Radio, and Inspectrum. SDR++ can read 32 bit floats or 16 bit integers, but not 64 bit floats.

What platforms use 64 bit floats as the default for complex out of curiosity?

nocarryr, 2023-12-26:
Since Python internally uses 64-bit precision for float, numpy [follows that behavior](https://numpy.org/doc/stable/user/basics.types.html#extended-precision).  The same is true for integers (Python uses something like `long long` I believe).

The type promotion rules can be found using [np.result_type](https://numpy.org/doc/stable/reference/generated/numpy.result_type.html) and are documented in [NEP 50](https://numpy.org/neps/nep-0050-scalar-promotion.html)

Here are some examples with typical numpy routines:
```python
>>> import numpy as np

>>> # default int is 64-bit
>>> np.arange(10).dtype
dtype('int64')

>>> # float operations convert it to float64
>>> (np.arange(10) * .1).dtype
dtype('float64')

>>> # complex literals convert it to complex128
>>> (np.arange(10) * 1-1j).dtype
dtype('complex128')

>>> # even when multiplied by a float32
>>> (np.arange(10) * np.zeros(10, dtype=np.float32)).dtype
dtype('float64')

>>> # unless it's against an int16 array
>>> (np.arange(10, dtype=np.int16) * np.zeros(10, dtype=np.float32)).dtype
dtype('float32')

>>> # default for complex (builtin type) is 128-bit
>>> np.zeros(10, dtype=complex).dtype
dtype('complex128')
```

And here are some that operate on an explicitly-typed complex64:
```python
>>> a = np.zeros(10, dtype=np.complex64)

>>> # operations with float literals retain the original dtype
>>> (a * 0.1).dtype
dtype('complex64')

>>> # same with complex literals
>>> (a * 1-1j).dtype
dtype('complex64')

>>> # but operations with other arrays (without dtype argument) promotes it to 128
>>> t = np.linspace(0, 1, 10)
>>> (a * t).dtype
dtype('complex128')

>>> # it only stays complex64 if ALL arrays in the operation are constrained
>>> t = np.linspace(0, 1, 10, dtype=np.float32)
>>> (a * t).dtype
dtype('complex64')
```

**NOTE**
I'm not trying to make an argument against using `complex64` and it would be trivial to implement.  Just trying to show that it wouldn't stay 64-bit without a bit of care taken by the user.  Documenting the above behavior would be a good idea I believe.